jerryscript-project/jerryscript #964: Re-thinking the Jerry Port API (proposal)

akiss77, 2016-03-12:
# Motivation

Lately, there have been several PRs (#945 #946 #956 #957) that pointed out that the port API of jerry is not that easy to use by embedders that are not in the main code base and thus could benefit from improvements.

Taking a closer look at some parts of the code also revealed that even the use cases in the main code base could become cleaner after some rethinking-refactoring.

# Analysis

I've built jerry with `make LTO=OFF ALL_IN_ONE=ON` and ran `nm` on the resulting jerry-core library to see what symbols it defines and -- most importantly -- what external symbols it references. Below is the list of external symbols (categorization by me):

```
"termination"
                 U abort
                 U exit
"i/o"
                 U printf
                 U putchar
                 U stderr
                 U vfprintf
"math"
                 U acos
                 U asin
                 U atan
                 U atan2
                 U ceil
                 U cos
                 U exp
                 U fabs
                 U floor
                 U fmod
                 U log
                 U pow
                 U sin
                 U sqrt
                 U tan
"general libc"
                 U longjmp
                 U memcmp
                 U memcpy
                 U memmove
                 U memset
                 U rand
                 U setjmp
                 U strlen
                 U strncmp
```

IMO, the "math" and "general libc" categories cover very fundamental functionalities, which a port shouldn't be allowed to override. However, the "termination" and "i/o" categories can be port specific, and we should think about what we actually use them for and how we should let the ports give their specific implementation.

# Proposal

I propose not to replicate the low-level libc API in the port API but to make the port functions "semantic".

## Port API: Termination

```c
/**
 * Signal the port that jerry experienced a fatal failure from which it cannot
 * recover.
 *
 * @param code gives the cause of the error.
 *
 * Note: jerry expects the function not to return.
 *
 * Example: a libc-based port may implement this with exit() or abort().
 */
void jerry_port_fatal (jerry_fatal_code_t code);
```

It should be the embedder application and/or the port that decides whether exit or abort is to be called. This differs from the current approach, where jerry even provides an api function to switch between exit/abort mode of fatal termination. This would make that function unnecessary.

_Note:_ It is highly questionable whether a library should be able to terminate an application. However, as of now, we only have the concept of completion code around jerry_parse and jerry_run. Most of the other API functions have no way of signaling an error. So, let's keep the termination approach for now with this port function.

## Port API: I/O

```c
/**
 * Print a string to the console. The function should implement a printf-like
 * interface, where the first argument specifies a format string on how to
 * stringify the rest of the parameter list.
 *
 * This function is only called with strings coming from the executed ECMAScript
 * wanting to print something as the result of its normal operation.
 *
 * Example: a libc-based port may implement this with vprintf().
 */
void jerry_port_log_console (const char *fmt, ...);

/**
 * Display or log an error message. The function should implement a printf-like
 * interface, where the first argument specifies a format string on how to
 * stringify the rest of the parameter list.
 *
 * This function is only called with messages coming from the jerry engine as
 * the result of some abnormal operation.
 *
 * @param level ??? (warning, error, fatal)
 *
 * Example: a libc-based port may implement this with vfprintf(stderr).
 */
void jerry_port_log_error (int level, const char *fmt, ...);

/**
 * Display or log a debug message. The function should implement a printf-like
 * interface, where the first argument specifies a format string on how to
 * stringify the rest of the parameter list.
 *
 * This function is only called with debug messages coming from the jerry engine
 * describing its internal operations (e.g., data structure dumps or tracing
 * info).
 *
 * @param level ??? (numeric?)
 *
 * Example: a libc-based port may implement this with vfprintf(logfile).
 */
void jerry_port_log_debug (int level, const char *fmt, ...);
```

These should be the only I/O functions jerry calls. All calls to putchar, vfprintf, and putchar should be rewritten to one of these, depending on what jerry wants to print.

It should be the port that decides what a console is, or whether error and debug messages are logged to the same console or saved to a database or to a file.

## Additional Port APIs

The above proposal covers and re-thinks only those functionalities, which are already in the project. There might be need for additional port APIs (e.g., memory management, #946). They should/might be analysed and proposed a similar way.

franc0is, 2016-03-14:
This proposal addresses a lot of the issues I have been raising in my pull requests, I hope it can be accepted and implemented quickly.

I am happy to re-purpose https://github.com/Samsung/jerryscript/pull/957 to implement `jerry_port_fatal`.

Longer term, I would like to see as much of Jerryscript's state as possible contained within a few data structures allocated by the port instead of the current static variables that are peppered throughout the code (and sometimes [aren't initialized properly](https://github.com/Samsung/jerryscript/pull/950)). I started moving in that direction with #946 and expect to spend more time on similar efforts in the future.

zherczeg, 2016-03-16:
I have no objection against this proposal.

akiss77, 2016-03-16:
3 in favor, none against. That's a good start. Any alternative suggestions for the function names or are they OK as proposed above? Any suggestions for the error/debug levels? Enums or ints? Or enum for the error level but int for the debug level? (There are some question marks in the proposal, I'd like to have some input there.)

LaszloLango, 2016-03-16:
IMO, `jerry_port_log_error` don't need level parameter. Currently we use 3 debug levels and I think that is enough. Enum for level parameter would be nicer.

franc0is, 2016-03-16:
FWIW, I think having both special purpose functions (`log_error`, `log_debug`) and levels is redundant. Instead, I'd suggest
`jerry_port_log(level, fmt, ...)` with
```
typedef enum {
  LOG_LEVEL_VERBOSE = 1,
  LOG_LEVEL_DEBUG = 10,
  LOG_LEVEL_INFO = 20,
  LOG_LEVEL_WARNING = 30,
  LOG_LEVEL_ERROR = 40,
  LOG_LEVEL_FATAL = 100
} LogLevel;
```

LaszloLango, 2016-03-17:
If we decide to use one port function to logging, then I suggest the following levels:

```C++
typedef enum
{
  JERRY_LOG_LEVEL_QUIET = 0, /* No message */
  JERRY_LOG_LEVEL_ERROR,     /* Default: only error messages */
  JERRY_LOG_LEVEL_DEBUG,     /* 1st level debug messages */
  JERRY_LOG_LEVEL_DDEBUG,    /* 2nd level debug messages */
  JERRY_LOG_LEVEL_DDDEBUG,   /* 3rd level debug messages */
} jerry_log_level_t;
```

franc0is, 2016-03-17:
Works for me :). My main point is that a "level" and distinct "error" and
"debug" functions is somewhat redundant.

akiss77, 2016-03-17:
I got to agree with @franc0is that a single `jerry_port_log (level, fmt, ...)` function will be enough. (Then, the "console" function would also better be suited with the name `jerry_port_console`.) And I also like level names suggested by him - or something similar. They are more aligned with existing logging systems, and their name is also more descriptive. (It's hard to deduce what the difference between 1st, 2nd, and 3rd level debug messages is.) Still, whatever names we choose, there should be some doc comments giving some hints, like:
* ERROR: jerry is in distress, it will terminate / bail out
* WARNING: jerry couldn't fulfill everything that was asked but has some fallback mechanism to continue execution (perhaps with limited functionality), e.g., memory stats were asked but related code is not built into the binary, JS execution is still possible
* DEBUG: info about jerry's internals, low volume, normally non-repeating, one-shot, e.g., literal storage contents at the end of JS execution
* TRACE: detailed info about jerry's internals, high volume, e.g., function enty/exit logs, or even looped prints

(I'm not aware of situations where we could differentiate between fatal and non-fatal errors -- at least for now --, thus listed only one ERROR level. I don't think that we have a valid use for INFO level: in most logging engines, INFO is something that shows normal progress of the app and is to be printed to the console; but we shouldn't print anything there unless it's either coming directly from the executed JS or signaling an error. For debugging, we are currently using only 2 levels -- not necessarily in consequent ways -- so something like DEBUG and TRACE might be enough for now.)

seanshpark, 2016-03-18:
@akiss77 , for me, `FILE*` had (so not currently not using) and `vprintf`, `vsprintf` has some problem linking in ES8266 port, so currently `jerry_port_errormsg()` doesn't work correctly in that port.
I don't know how but somehow it would be nice not using `var_list` for `...` .

LaszloLango, 2016-03-18:
Related PR: #969
What about moving all date related functions to jerry port?

akiss77, 2016-03-18:
@seanshpark Do `FILE`, `vXprintf`, varargs work in general, in an example code? Do you experience the issue with the jerry port only? Or is it a problem everywhere on ES8266? (I'm worried because varargs is a language feature, it should work everywhere.)

zherczeg, 2016-03-18:
I also have a proposal. I saw several PRs are opened with @franc0is. It would be great to land many of them, but I am confused how they exactly affect the project. Unfortunately touching the API is always a complex and time consuming task. A good API must cover as many use cases as possible, efficiently. I am aware that we need to improve the Jerry API, but I hoped we will do it later :) But things are changed.

Here is my proposal then: before landing actual changes, we should move the design plan from here into a wiki page. When we agree on a certain part, we should update that wiki page (so everybody can see the last version of the planned API). In this discussion we should just focus only on the pure interface. When we discussed it fully, we should start implementing it.

seanshpark, 2016-03-18:
@akiss77 , I had link problem with ESP8266 with `FILE*`(`stdin`, `stdout`, `stderr`). It had some  internal implementation that I could not access in the link. I may have been doing something wrong, but surely was strange.

+1 to @zherczeg :)

akiss77, 2016-03-18:
@seanshpark, You've also mentioned a varags issue in you comment. Is that still valid and/or related?

franc0is, 2016-03-18:
I fully agree with @zherczeg on the way forward. My PRs were meant as a conversation starter and because I would like my fork not to diverge from upstream as much as possible (I made these changes out of necessity originally).

In addition to the the "termination" and "logging" APIs above, I can foresee needing a couple more:
* A "datetime" port API (see #969) which would provide a `_get_time` and a `_get_tzinfo` APIs. Jerryscript currently relies on gettimeofday which is a Unix API and thus not very portable.

* Memory management APIs (see #946) which would be used to allocate & manage memory used to keep the state of the engine. Of immediate interest to me is finer grained control over where the `mem-heap` ends up being stored. So that (1) I do not have to incur the memory penalty of the heap when the engine is not running and (2) I can allocate this large chunk of RAM wherever is most convenient. Most embedded systems have several special purpose allocation schemes, let the port figure out which makes the most sense to use rather than impose static allocation.

In the future, I would be interested to discuss how the port APIs could be leveraged to manage the *state* of the engine. Currently, heavy use of static variables makes running more than one instance of the engine impractical, and makes engine initialization error prone. It's also a waste of memory when the engine isn't being used. Such a change would have wide-ranging implications, and would probably require changes to the embedding APIs as well, so I think we should treat it as a separate item.

seanshpark, 2016-03-20:
@akiss77 , I think "varags issue" was about  some problem with using "vsprintf" in compile time.

zherczeg, 2016-03-21:
I started to create a wiki page from the discussion here:

https://github.com/Samsung/jerryscript/wiki/Current-version-of-the-Jerry-Port-API

Feel free to add your suggestions. I would like to hear @LaszloLango opinion about the Date functions.

Regarding contexts, I think there are 3 use cases which should be satisfied:
- completely static engine for ultra low-end devices. One engine instance, static global heap.
- static engine, with optional initialization / finalization (requested by @franc0is, I am not sure I fully understand this)
- multi-engine mode where each thread can maintain multiple engines (cross-thread use of the same engine is still not supported)

LaszloLango, 2016-03-21:
I'd propose the following API for date:

```C++
/**
 * Jerry time structure
 */
struct
{
  long int tv_sec;   /* seconds since Unix epoch */
  int tz_minutewest; /* minutes from west */
  int tz_dsttime;    /* daylight saving time */
} jerry_time_t;

/**
 * Get system time, timezone and daylight saving data
 *
 * @return true  - if success
 *         false - otherwise
 */
bool jerry_port_get_time (jerry_time_t *);
```

franc0is, 2016-03-21:
@zherczeg let me give you more details on my initialization/finalization needs.

The Pebble firmware features several heaps. One of those heaps is dedicated to application memory and is not memory protected, all other heaps are memory protected.

I want to be able to allocate the JerryScript heap inside of this application heap when an application is launched which wants to execute some javascript. That way, an application which does not execute javascript can use that memory for something else, and the memory available to the kernel does not change whether or not javascript is used.

Does that make more sense?

franc0is, 2016-03-21:
@LaszloLango your `jerry_time_t` needs to include the number of milliseconds since the last second as well. 

I would prefer `tz_minutewest` to be `tz_offset_minutes`. Perhaps `tv_sec` could be more explicitly named as well...

zherczeg, 2016-03-22:
@franc0is yes, your request is not about multiple heaps, but about using a custom heap. You want to save all Jerry global memory when Jerry is not used. Reasonable request.

In that case tasks are:
- Supporting multiple heaps using context pointers (optional, compile time switch)
- Allocating heap(s) for the application

My first suggestions:

When multiple heaps are supported, there should be a function which tells the memory required for a given Jerry heap

    size_t jerry_context_size_for_heap(size_t heap_size_in_kbyte);
    // size of area pointed ptr must be at least jerry_context_size_for_heap size
    void jerry_init_heap(void *ptr);
    void jerry_free_heap(void *ptr);

I think for the second I would maintain a global variable for pointing that heap rather than callbacks. One pointer should not be a problem for any application.

akiss77, 2016-03-23:
@LaszloLango @franc0is My two cents on the Time Port API: I'd suggest changing the names of the struct. (At the very least.) In the way they are currently proposed, they follow libc/POSIX terminology closely. The goal of the port API(s) should not be to duplicate the libc API again, but to specify an API for the needs of jerry.

To be less abstract, an example: I deliberately _did not_ propose `jerry_port_abort` and `jerry_port_exit` functions for the Termination API. That would have been very libc specific. Yes, the current implementation in jrt-fatal.c does use both `abort` and `exit` and implements a logic to choose between the too -- but that's exactly what we want to replace and leave for the port to implement. The need of jerry is to quit with a given error code. Whether `jerry_port_fatal` is implemented with `exit`, `abort`, or something else, becomes unimportant for jerry.

Thus, we should think (or re-think) whether `tv_sec`, `tz_minutewest`, and `tz_dsttime` are really the data we need from the port, and whether those names describe them the best. We are free/we ought to align them to our needs and terminology. The `tv_` and `tz_` prefixes are bad smells to me in that sense. (To come up with libc-independent jerry-specific structs and names needs the investigation of those parts of the code that deal with date/time related information.)

franc0is, 2016-03-25:
@akiss77 we're on the same page, which is why I suggested we not use `tz_minutewest` and `tv_sec`.

LaszloLango, 2016-03-25:
You are right. The date API should be based on ECMA-262 standard and not on libc/POSIX.
Related sections in the standards:
http://www.ecma-international.org/ecma-262/5.1/index.html#sec-15.9.1.7
http://www.ecma-international.org/ecma-262/5.1/index.html#sec-15.9.1.8
http://www.ecma-international.org/ecma-262/5.1/index.html#sec-15.9.4.4

How about the following API:
```C++
/**
 * Get system time
 *
 * @return milliseconds since Unix epoch
 */
unsigned long int jerry_port_get_current_time ();

/**
 * Get local time zone adjustment
 *
 * @return minutes from west
 */
unsigned int jerry_port_get_local_time_zone_adjustment ();

/**
 * Get daylight saving time adjustment
 *
 * @return daylight saving time adjustment
 */
unsigned long int jerry_port_get_current_time (unsigned long int time);
```

OR

```C++
/**
 * Jerry time structure
 */
struct
{
  unsigned long int millisec_since_epoch; /* milliseconds since Unix epoch */
  int time_zone_offset; /* minutes from west */
  int time_zone_daylight_saving_time; /* daylight saving time */
} jerry_time_t;

/**
 * Get system time, timezone and daylight saving data
 *
 * @return true  - if success
 *         false - otherwise
 */
bool jerry_port_get_time (jerry_time_t *);
```

zherczeg, 2016-03-25:
I would prefer the second one.

akiss77, 2016-03-25:
I'd say, it depends on how the results are used. Are they always used together? Then they should go in one struct and returned by a single function. Do different call sites need different date/time aspects? Then we should have multiple port functions.

LaszloLango, 2016-03-25:
@akiss77, timezone and daylightsaving are always used together. The current time is not.

akiss77, 2016-04-05:
@LaszloLango @zherczeg The discussion got stalled, it seems. Will there be another iteration of the Date Port API?

franc0is, 2016-04-05:
Let me know how I can help. At the moment, I cannot sync my branch with upstream until the memory management bits are figured out.

LaszloLango, 2016-04-06:
@akiss77, based on you comment I suggest a 3rd version.

@akiss77, @zherczeg, @franc0is, I'm trying to summarize the date API suggestions. Please tell me your opinion.

#### 1st

```C++
/**
 * Get system time
 *
 * @return milliseconds since Unix epoch
 */
unsigned long int jerry_port_get_current_time ();

/**
 * Get local time zone adjustment
 *
 * @return minutes from west
 */
unsigned int jerry_port_get_local_time_zone_adjustment ();

/**
 * Get daylight saving time adjustment
 *
 * @return daylight saving time adjustment
 */
unsigned long int jerry_port_get_current_time (unsigned long int time);
```

#### 2nd

```C++
/**
 * Jerry time structure
 */
struct
{
  unsigned long int millisec_since_epoch; /* milliseconds since Unix epoch */
  int time_zone_offset; /* minutes from west */
  int time_zone_daylight_saving_time; /* daylight saving time */
} jerry_time_t;

/**
 * Get system time, timezone and daylight saving data
 *
 * @return true  - if success
 *         false - otherwise
 */
bool jerry_port_get_time (jerry_time_t *);
```

#### 3rd

```C++
/**
 * Jerry time zone structure
 */
struct
{
  int time_zone_offset; /* minutes from west */
  int time_zone_daylight_saving_time; /* daylight saving time */
} jerry_time_zone_t;

/**
 * Get timezone and daylight saving data
 *
 * @return true  - if success
 *         false - otherwise
 */
bool jerry_port_get_time (jerry_time_zone_t *);

/**
 * Get system time
 *
 * @return milliseconds since Unix epoch
 */
unsigned long int jerry_port_get_time ();
```

I think 3rd would be the best.

akiss77, 2016-04-06:
@LaszloLango Thanks, great! I'd also vote for the 3rd option. With the minor comment that the two functions share the same name right now. What about `jerry_port_get_time_zone` and `jerry_port_get_current_time` ?

LaszloLango, 2016-04-06:
@akiss77, fixed :)

zherczeg, 2016-04-06:
I vote for 3rd as well. Please update the wiki.

akiss77, 2016-04-06:
Wiki updated. Has some structure now and some errors got fixed.

LaszloLango, 2016-04-07:
@akiss77, I'm not sure about the I/O part. Why do we need 2 function for logging? What is the difference between `jerry_port_console` and `jerry_port_log`? I don't like `jerry_log_level_t`. From the names it is like flags, not levels.

zherczeg, 2016-04-07:
Me too, we only need one logging function. I like the level, since people usually configure the logging level than a list of enable/disable for each flags.

akiss77, 2016-04-07:
@LaszloLango @zherczeg The discussion was always about 3 or 2 I/O functions (see above). Actually, the original proposal was 3 (by me), then @franc0is pointed out that the  two logging functions could be merged into one. However, when it got copied to the wiki, only one remained. That's what I fixed.

The why: actually, it is in the doc comments of the two functions. There should be a function that provides the executed JS code a way to *print* something to the "console", whatever the concept of console on a given port is. This printing is not initiated by the engine directly. However, there are cases when the engine wants to *log* some diagnostics of a given severity. This is not necessarily something that should be mingled with the console (but can be). Logging and printing are two different concepts: printing is a normal and unsuppressable activity of an application, while logging is for tracking the activity of an application. Both may result in some text appearing on the same stream/screen, but that's not for us to decide. IMHO.

zherczeg, 2016-04-07:
I disagree with @akiss77. If we allow a simple print function, people will use that always, and that can be a lot of unsupressable logs. See the Linux kernel, you always need to specify the severity to printk. We can add levels like USER_PRINT, USER_HINT.

akiss77, 2016-04-07:
I disagree with @zherczeg ... But `printk` and the kernel may by good analogies, I'll try to explain my point better with them. 

`printk` is a kernel logging routine. But it is not the only way to get something printed, actually, it is only for kernel diagnostics (which may go to console and/or to a log file, depending on settings). However, the kernel *does* expose syscalls to userland (`write` in this case) to get something written somewhere (i.e., printed to console, if the file descriptor is stdout). In this case, `printk` is *not* used. Finally, to make life easier and avoid the need of using syscalls directly, we have `printf` in libc (which has no "level" concept).

In this analogy, `jerry_port_log` is `printk`. Used only for diagnosing the engine, directing log messages "somewhere". However, `jerry_port_console` corresponds to the syscall that gives access to the console (always and only to the console). Finally, a `print` function bound into the JS execution environment (i.e., callable by JS code) is somewhat like the `printf` of libc.

Finally, about the people. It's us. It's discipline. It's reviews. (Moreover, should someone want to be sloppy, we cannot prevent her/him, since libc is always there and one might always be tempted to call `printf` directly, even for diagnostics. Only reviews can notice that and block it creeping into the code base.)

akiss77, 2016-04-08:
I've tried prototyping at least the Termination API as there seems to be no disagreement. However, I've run into some issues that should be discussed further as may have effect on the whole port concept. To facilitate that discussion, I've drawn an ASCII art component diagram of JerryScript and its connected libraries (and apps) -- at least how I understood and envisioned it:

```
+------------------------------------------------------------------------------+
|                                                                              |
|                             Embedder Application                             |
|                                                                              |
+--------------------------====================================================+

+---Port-specific API---+ +==================JerryScript API===================+
|                       | |                                                    |
|                       | |                  JerryScript Lib                   |
|                       | |                                                    |
|                       | +====================+                               |
|                       |                      |                               |
|                       +===Jerry Port API===+ |                               |
|                                            | |                               |
|   JerryScript Port Implementation Lib      | |                               |
|                                            | |                               |
+--------------------------------------------+ +===============================+

+--------------------------LibC API-------------==============+ +===LibM API===+
|                                                             | |              |
|                            C Lib                            | |   Math Lib   |
|                                                             | |              |
+-------------------------------------------------------------+ +--------------+
```

Legend:
* `===`: mandatory interface/dependency
* `---`: optional interface/dependency

That is, JerryScript (the lib) has 3 dependencies: it needs mathematical functions from a libm implementation (which can be jerry-libm for minimal code size, or a "regular" libm, e.g., from GNU); it needs some basic libc routines (e.g., rand, set/longjmp, mem*; which can come from jerry-libc, or again from a "regular libc implementation); and it needs some functions from a port implementation (which APIs we are defining right in this discussion).

What occured to me is that even the port implementation may have its own extra API in addition to the "official" JerryScript API. As an example: if we would like to move the current termination logic to the default port (i.e..: right now, we decide based on a flag whether to call `exit` or `abort` to quit), then we cannot keep that flag among the "jerry" flags, as this behaviour is specific to the default port. (Another port may want to call only one function on fatal conditions, which may be neither `abort` nor `exit`, but something entirely different. There, `JERRY_FLAG_ABORT_ON_FAIL` makes no sense.) However, then we need something like `jerry_port_default_set_abort_on_fail (bool)`, and this cannot be declared in the core jerry headers. (The issue of logging is something similar: it may/should be port specific how to control log level filtering -- not done within jerry-core. Thus, yet another port-specific function.) Which is not a bad thing, I'm saying, just have to take it into consideration. (Moreover, an embedder app does not *necessarily* need to use any port-specific APIs. The port implementation should be written to have sensible defaults for anything configurable.)

Another thing that I realized is that right now the port is built into jerry-core. It's a bit of gut feeling right now, but the component diagram above suggests me that the port could be built on its own (especially if it's not the default one but coming from an external source).

And my final observation: our `targets` directory kind of mingles the "port" and "embedder app" concepts. The `targets/{esp8266,mbedk64f,nuttx-stm32f4}` directories correspond to the embedder app box in the diagram (i.e., "above" the JerryScript lib), while `targets/default` is (the start of) a libc-based port implementation (i.e., "below" the JerryScript lib). It might be worth moving the latter out to `jerry-port/default`, I'd suggest.

Do I see things right? Anyone agrees? Disagrees?

seanshpark, 2016-04-11:
@akiss77 , how could it be presented in some directory name?

akiss77, 2016-04-11:
@seanshpark , I'm sorry, I'm not completely sure what "it" is referring to in the question.

seanshpark, 2016-04-11:
@akiss77 , what you explained about with a diagram above.

akiss77, 2016-04-11:
Well, something like:
* jerry-libc
  * include
    * *.h
  * (all our sources)
* jerry-libm
  * include
    * math.h
  * (all our sources)
* jerry-port
  * default
    * jerry-port-default.h (or this can go in its own include subdir as well, if we want a nice clean regular structure, as this is intended to be a public header)
    * fatal.c
    * io.c
    * (etc...)
* jerry-core
  * (all the lot we already have here)
* targets (although we may have better name for this as well, e.g., jerry-app(s), but it's not crucial to change right now)
  * (all the existing embedders, and perhaps main-unix.c and main-mcu.c moved uner this directory too; and these apps could use `#include "jerry.h"` and `#include "jerry-port-default.h"`, with the latter one being optional)

Most of the above we already have in place. Only my last paragraph was about changing the dir structure a bit -- and it was about suggesting the move of the port out of the targets dir.

seanshpark, 2016-04-11:
@akiss77 , thank you for the explanation. I can now imagine how it will be and I agree with your suggestion.

zherczeg, 2016-04-11:
Personally I don't like jerry-port and target names, I would change them to jerry-external-deps, and jerry-platforms

akiss77, 2016-04-11:
@zherczeg I'm not sure which name is for which component in your proposal? 

If "external-deps" is for "ports", then I'm not sure its a good fit. Actually, we need at least one port that's internal, that we maintain. Both as an example for others and also as something our own command line interpreter (main-unix.c) can use.

And I'm also not sure what "platform" stands for. To me, a platform is something I base my work on. And applications that embed jerry are based on jerry and not the other way around.

But I do agree that "targets" is not the best name we have right now. (However, I was happy with the "port" terminology as it is used in other projects as well which I'm familiar with [WK].)

franc0is, 2016-04-11:
@akiss77 I like your diagram and explanations a lot. A few thoughts however:
1. In my opinion, the embedder's application should not belong to this repository. I certainly have taken that approach for my work, and in a way you guys have too: the `iotjs` (i.e. application) code is outside of this repository
2. At some points you will need to separate the line between the "official" ports (e.g. sm32f4, esp, nuttx) and non-official ones. In the past, I've seen projects use a "contrib" repository for unofficial ports.
3. I don't think `printk` is a good analygy for the port printing APIs. I see the port as the interface to the host hardware. The hardware does not care whether you are printing a diagnostics message or an application message. For that reason, I think a single API is all that is necessary. That API exposes a "level" because it is a common need of all the different consumers of that API that messages can easily be filtered.

jiangzidong, 2016-06-28:
Hi, we met the same I/O issue when we trying to port jerry into a new platform. Jerry-core will call printf directly (such as in ecma-builtin-global.c and heap stats), while the printf has special behavior in the  platform, so we had to change the content of jerry-core. It will be great if jerry-core use port-api "jerry-port-console/log" instead of printf.

I agree the content in wiki page, and may I know when those changes will merged?

LaszloLango, 2016-06-28:
@jiangzidong, soon :)

martijnthe, 2016-08-05:
Re. "multi-engine mode":

I noticed the recent addition of [jcontext.h|c](https://github.com/Samsung/jerryscript/blob/master/jerry-core/jcontext/jcontext.h).

This is a nice step towards supporting the "multi-engine mode" / creating the engine context dynamically, but as-is, I don't think it will support that use case yet.

Or is the idea that one would need to redefine `JERRY_CONTEXT` / `JERRY_HEAP_CONTEXT` / `JERRY_HASH_TABLE_CONTEXT`?

What's the latest thinking and plan to support multiple JerryScript engine instances?

zherczeg, 2016-08-05:
Exactly.

We (you) could add more context profiles, and exactly one must be selected when jerry is compiled. The current is the "default context profile", and will be the default when no other profile is selected. The code should looke like something like this:

    #ifdef JERRY_CONTEXT_PROFILE_name_of_the_profile1

    #define JERRY_CONTEXT(field) ...
    #define JERRY_HEAP_CONTEXT(field) ...
    #define JERRY_HASH_TABLE_CONTEXT(field) ...

    #elif defined JERRY_CONTEXT_PROFILE_name_of_the_profile2

    #define JERRY_CONTEXT(field) ...
    #define JERRY_HEAP_CONTEXT(field) ...
    #define JERRY_HASH_TABLE_CONTEXT(field) ...

    [...]

    #else

    #define JERRY_CONTEXT(field) ...
    #define JERRY_HEAP_CONTEXT(field) ...
    #define JERRY_HASH_TABLE_CONTEXT(field) ...

    #endif

Some profiles might add extra API / port API functions (only available when that profile is selected).

We might not want to add every specialized profile to the trunk, but we want an easy interface to create new profiles and maintain them by individuals.

akosthekiss, 2016-08-19:
The Termination, Date, and I/O port APIs are all in place, so this issue can be closed now. Should new port APIs be needed, they can be discussed further under new issues.

closed 2016-08-19 by akosthekiss