jerryscript-project/jerryscript #992: Nominating Akos Kiss (akiss77) for JerryScript Maintainer status

zherczeg, 2016-04-07:
Hi,

I would like to nominate Akos Kiss to become a JerrzScript maintainer. He already have reviewed 38 pull requests, and landed 56 patches. The list of these follows.

38 informal reviews:

* #971 : Optimize string character access for ascii strings 
* #970 : Fix precommit scripts on OSX 
* #961 : Use code unit instead of code point 
* #955 : Allow external builds to use non-gnu compilers 
* #949 : Use '-Werror' instead of '-Werror=...' for Jerry libc 
* #948 : Storing byte code size in the byte code header. 
* #947 : Few improvements for RegExp 
* #947 : Make target provide jerry-port.c 
* #944 : Add acquire_value API to simplify caller code 
* #941 : Don't use messages for errors by default 
* #938 : Warning fixes. 
* #931 : Fix warning when returning from vm_loop. 
* #929 : Disable date object related system calls by default. 
* #920 : Non-recursive vm_loop 
* #917 : Fix the format of doxygen comments of output arguments 
* #916 : Add another argument for the JERRY_STATIC_ASSERT with the description of the assert statement. 
* #915 : Build fix 
* #914 : Various enhancements 
* #912 : Fix 'jerry-test-suite/13/13-011.js' fail in compact profile 
* #904 : Rearrange fields of ecma_property_t to be naturally aligned. 
* #901 : Remove EXTERN_C macros and use block based solution 
* #896 : Fix style issues and improve vera++ rules. 
* #893 : Optimize 'lit_get_magic_string_size' calls 
* #888 : Remove completion value. 
* #885 : Warning fixes. 
* #884 : README.md: add an hello.js example 
* #877 : Improve 'tools/run-perf-test.sh' 
* #876 : Eliminate cpp featureset 
* #873 : Fix system call related date builtin functions 
* #868 : Move the logic of Valgrind-support check to the CMake part. 
* #866 : Abort the build-process when the build-configuration is unsupported 
* #852 : Fix 'Segmentation fault on String.Replace' 
* #838 : Fix the correct number of command line arguments check. 
* #836 : remove the c++ syntax, struct::xxx 
* #831 : Add JerryScript logo. 
* #813 : Compact Byte Code parser and executor for Jerry (release branch). 
* #790 : Test the build options 
* #488 : Enhancement of parser operand handling, compaction of instruction dumper 

56 pull requests merged into master (touching the build system, tests, online testing infrastructure, libc & fdlibm fixes, refactorings, fixes, and code size improvements in core):

* #974 : Fix regexp character classes with \uNNNN and \xNN code points
* #973 : Fix: code-related vera++ rules should not be enforced in comments
* #972 : Remove exec mode from non-executable files
* #967 : Add unit testing of fdlibm
* #965 : Fix SCCS IDs in fdlibm
* #963 : Fix `.size longjmp` in the linux port of jerry-libc
* #960 : Refactor the printing of memory usage statistics
* #959 : Improve the dumping of literals
* #958 : Fix initialization from incompatible pointer type error in `re_dump_bytecode`
* #954 : Remove support for per-instruction memory statistics
* #953 : Fix terminology of snapshot saving in literals
* #952 : Unify the naming scheme of all header defines
* #951 : Check unit tests with vera++ & fix reported issues
* #943 : Fix size types
* #937 : Remove dead macros from memory management code
* #936 : Change pool-based allocation to heap-based allocation in jerry.c
* #934 : Remove memory management-related dead code
* #932 : Add full profile to MCU build targets
* #928 : Add `STRIP_RELEASE_BINARY` option to Makefile
* #926 : Remove the prerequisites installation and MCU build steps from Travis CI
* #925 : Remove default handling of build options from Makefile
* #924 : Remove unused `EXTERNAL_` variables from Makefile
* #913 : Replace bit field manipulation functions with macros
* #909 : Improve the usability of test runner scripts
* #908 : Replace the `EXTERN_C` macro with `extern "C" {` block in fdlibm
* #907 : Drop unimplemented run context stubs
* #906 : Add some more `const`s to get data moved from .data to .rodata
* #905 : Refactor ECMA builtin template
* #903 : Move cppcheck logic from Makefile and CMakeLists.txt to tools/check-cppcheck.sh
* #902 : Fix tools/check-signed-off.sh to handle PRs not based on HEAD
* #899 : Add merge commit support to tools/check-signed-off.sh and…
* #898 : Add ARM cross build and QEMU user space emulated tests to the CI
* #895 : Quick fix for Travis CI: remove signed-off check
* #894 : Add Travis CI badge to readme
* #892 : Turn some global variables `static const`
* #891 : Get the alpha order of lexer keywords right
* #890 : Change the layout of the doxygen-generated documentation
* #889 : Eliminating doxygen warnings by fixing the documentation documentation enhancement
* #887 : Move Doxyfile to the root directory of the project enhancement infrastructure
* #886 : Add support for travis-ci integration enhancement infrastructure
* #880 : Remove `jerry_port_` functions from test-common.h bug minor
* #872 : Remove cppcheck and vera++ from prerequisites enhancement infrastructure style
* #871 : Remove references to non-existent parser/js/bc and parser/js/collections directories enhancement infrastructure
* #870 : Rename docs/Doxygen to docs/Doxyfile enhancement infrastructure
* #867 : Refactoring the build system enhancement infrastructure
* #849 : Add generated documentation to gitignore enhancement infrastructure
* #845 : Refactoring jerry-libc to squeeze out some extra bytes from the code (+fixes)
* #841 : Eliminate code duplication in memory statistics printing enhancement memory management
* #834 : Revive mem_stats.sh
* #753 : Tab removal
* #719 : Pass -Wno-error to linker in case of LTO builds
* #715 : Add a "Hello JerryScript!" program to the tests
* #408 : Makefile refactor
* #146 : Make exit behaviour of `jerry_fatal` flag-dependent
* #141 : Add `abort ()` to jerry-libc
* #134 : Terminate with signal in case of an assert

Any seconds?

LaszloLango, 2016-05-05:
Ákos got 5 seconds and no objection in the last 4 weeks. So we can close the nomination process. I added him to the maintainers list. Congratulations! :)

closed 2016-05-05 by LaszloLango

thegecko, 2016-05-05:
nice one @akiss77