jerryscript-project/jerryscript #1027: Build Jerryscript for LLVM

ssultana, 2016-04-22:
Hi,
Previously I [built Jerryscript for Contiki OS/Galileo board](https://github.com/Samsung/jerryscript/issues/827). Basically, I  made a directory `jerryscript/targets/contiki-x86`, wrote a Makefile `jerryscript/targets/contiki-x86/Makefile.contikix86` for Contiki/Galileo board and set the appropriate toolchain. Now, I want to build Jerryscript for LLVM in Contiki/Galileo board. Can you please point me to what Makefile/compiler options I should change? So far I tried the following changes but haven't been able to build JerryScript successfully.

1. Modify the compiler options in  Makefile.contikix86 

	```
jerry:
        ... ... 
	 -DCMAKE_TOOLCHAIN_FILE=build/configs/toolchain_external_clang.cmake \
	 -DEXTERNAL_LIBC_INTERFACE=$(CONTIKI)/platform/galileo/bsp/libc/i586-elf/include \
	 -DEXTERNAL_CMAKE_SYSTEM_PROCESSOR=x86 \
	 -DEXTERNAL_CMAKE_C_COMPILER=clang \
	 -DEXTERNAL_CMAKE_CXX_COMPILER=clang++ \
	 -DEXTERNAL_COMPILE_FLAGS="$(EXT_CFLAGS) $(CFLAGS)"
```

2. Write `build/configs/toolchain_external_clang.cmake` as
     
      ```
      ... ... 
      CMAKE_FORCE_C_COMPILER(${EXTERNAL_CMAKE_C_COMPILER} Clang)
      CMAKE_FORCE_CXX_COMPILER(${EXTERNAL_CMAKE_CXX_COMPILER} Clang)
```
3. Changed Imported Library Options in ` JerryScript/CMakeLists.txt` as
```
elseif("${CMAKE_C_COMPILER_ID}" STREQUAL "Clang")
  add_library(${PREFIX_IMPORTED_LIB}libclang STATIC IMPORTED)
  execute_process(COMMAND ${CMAKE_C_COMPILER} ${FLAGS_COMMON_ARCH} -print-file-name=
                  OUTPUT_VARIABLE IMPORTED_LIBCLANG_RT_LOCATION
                  OUTPUT_STRIP_TRAILING_WHITESPACE)
  set(IMPORTED_LIBCLANG_RT_LOCATION "/home/salmin/Documents/llvm-git/build/lib/libclang.so")
  set_property(TARGET ${PREFIX_IMPORTED_LIB}libclang
               PROPERTY IMPORTED_LOCATION ${IMPORTED_LIBCLANG_RT_LOCATION})
```

seanshpark, 2016-04-25:
@ssultana , Hello,
I've not much experience with LLVM Clang, but it seems that some compile options are different. With your comment above I don't see any problem. If possible, can you post compile or link errors here?
May be you already have tried like this (http://stackoverflow.com/questions/7031126/switching-between-gcc-and-clang-llvm-using-cmake) page...

tilmannOSG, 2016-07-27:
Please have a look at #1192, with the the new build system it is very easy to build with Clang.

closed 2016-07-28 by zherczeg

ssultana, 2016-07-28:
Thanks @tilmannOSG . I'll check that.