jerryscript-project/jerryscript #1032: _gettimeofday linking error for target-mbed

thegecko, 2016-04-26:
Since commit https://github.com/Samsung/jerryscript/commit/b523cf3cd10c02d96cfa43b4469508eabd58cf52

The ```k64f``` target which can be compiled with ```target-mbed``` in PR https://github.com/Samsung/jerryscript/pull/1019

Fails with:
```bash
/usr/bin/../lib/gcc/arm-none-eabi/5.2.1/../../../../arm-none-eabi/lib/armv7e-m/libc_nano.a(lib_a-gettimeofdayr.o): In function `_gettimeofday_r':
gettimeofdayr.c:(.text._gettimeofday_r+0xe): undefined reference to `_gettimeofday'
```

@akiss77 @knightburton

galpeter, 2016-04-26:
I think I've found the issue(s):
1) the `EXTERNAL_PORT_DIR` is not set when building for the mbed target, thus the default date-time port implementation gets used.
2) But the default date-time implementation calls the `gettimeofday` and for that it seems the arm requires a _gettimeofday implementation (which makes sense in embedded context imho).

So what we need to do is simply set the `EXTERNAL_PORT_DIR` and implement the required date-time port api.

knightburton, 2016-04-29:
I've committed a solution for this, please check the PR again.

akiss77, 2016-05-06:
Fixed by #1019 (AFAIK)

closed 2016-05-06 by akiss77