jerryscript-project/jerryscript #1052: Namespace for internal functions

stephenkyle-ARM, 2016-05-11:
If jerryscript is intended to be built as an external library for inclusion into other projects, are there any thoughts on renaming all internal functions (for example, 'mem_init') that may clash with functions in other libraries, to give them a namespace?

LaszloLango, 2016-05-11:
We are tying to use `jerry_` or `ecma_` prefix everywhere. I think adding a prefix to the conflicting functions could be a good solution for the problem. IMHO, the whole jerry-core/mem directory should use `jerry_mem_` prefix instead of `mem_`.

zherczeg, 2016-05-12:
Yes, mem_ is the namespace. If we want to change that we should change the directory name and change all prefixes.

stephenkyle-ARM, 2016-05-13:
Thanks for the responses. I'd appreciate if the namespace could be changed for the mem functions, as I currently cannot build a binary with both jerryscript and a lwIP stack that I'm using, as I'm encountering this exact function name clash between them.

tilmannOSG, 2016-06-15:
Looks like this has been fixed, can we close this issue?

tilmannOSG, 2016-06-15:
Accidentally closed it, please confirm :)

zherczeg, 2016-06-15:
I hope so

LaszloLango, 2016-06-16:
We can close this issue. The `mem_init` name clash is solved. If there will other problems with the current function namespaces, then we can raise a new issue for them.

closed 2016-06-16 by LaszloLango