jerryscript-project/jerryscript #1067: mbed target script importing

stephenkyle-ARM, 2016-05-18:
Hi,

At the moment in the mbed target build, the behaviour when loading JS files is to execute main.js first, and then evaluate all the other js files (in whichever order they were turned into C files by js2c). This can leave some applications rather fragile, if their execution depends on the order in which their scripts are evaluated.

Is there a plan to introduce a function that allows explicit inclusion of files like 'require()' to this jerryscript target, or is that more intended for iot.js? Thanks.

zherczeg, 2016-05-24:
Normally functions like require() is implemented in higher abstraction levels since concepts such as file system is not part of EcmaScript 5.1 and JerryScript does not depend on it. However it is easy to add such a function using the JerryScript API. Is that acceptable for you?

stephenkyle-ARM, 2016-05-24:
Hi. This was more intended to open up a discussion about what you think would be the best way to go about doing this? Currently js2c populates an array called js_codes with all .js files found in js/, and these are all automatically loaded on startup. Instead if we added a new native function called, for example, require/import, this could then find the correct script in the array, and eval it at that location. The only difference to node.js in that case, is that you would add everything to the global namespace, and not capture it inside an object.

zherczeg, 2016-05-24:
Yes, with eval everything uses the same global object (lexical environment). Perhaps a strict eval could help to separate the global and local contexts.

zherczeg, 2016-05-24:
eval("var a = 5"); print(a); -> prints 5
    'use strict'; eval("var a = 5"); print(a); -> prints ReferenceError: a is not defined

stephenkyle-ARM, 2016-05-24:
Right, but usually you want to include something to export some values to the global scope, i.e., functions you may want to call in a library module you're loading. Won't use strict prevent that entirely?

zherczeg, 2016-05-24:
It is a bit more inconvenient but you can do this:

    'use strict'; eval("var a = 5; this.b = 6"); print(b /* ok */); print(a /* ref error */);

zherczeg, 2016-05-24:
A (new Function())("str") could also help, but it would allow "return" statements in the script. If that is not a problem, you can also use that.

Perhaps a custom eval could help solving this problem. It creates a lexical environment like in strict mode, but the script is not forced to strict mode conformance.

stephenkyle-ARM, 2016-05-24:
A custom eval would certainly help if it could return the lexical environment captured in an object, so I can essentially do something like

colorsensor = require('colorsensor')

And then have colorsensor contain all declared variables from the lexical environment of 'colorsensor'. Perhaps we should go one further and have the custom eval only return things added to a particular object, as module.exports works in node.js.

thegecko, 2016-05-24:
Following an asynchronous dependency standard like [CommonJS](http://wiki.commonjs.org/wiki/CommonJS) (require/exports) as used in node.js would be very familiar to existing JavaScript developers.

Another format to consider is [ES2015 modules](http://www.ecma-international.org/ecma-262/6.0/#sec-modules) (import/export) as this is a web standard which is becoming popular and feels more future-proof.

akosthekiss, 2016-08-19:
This issue seems to have been abandoned for a while. Still, my two cents: I agree with @zherczeg that `require` and similar functions belong to higher abstraction levels. JerryScript is an engine and right now it is aiming at ES5.1 only (but is quite good in that regard, IMHO). It's the responsibility of its embedders to add similar functionalities. One such embedder is IoT.js. (Note: when node.js has been mentioned above multiple times, it became clear that it's not JerryScript that has to provide `require`. As in that case it's not the V8 JS engine that does it either.) Another could be the mbed target app. But honestly, I think that it would be an overkill for the mbed port to implement anything like that.

To be constructive: if all you need is to ensure that some predefined JS scripts get evaluated in a given order, then you could do that in the Linux-ish `/etc/init.d` way: prefix everything with a one or two-digit number and ensure that js2c.py keeps things sorted. This might even be helpful for other targets. (Alternatively: the mbed target directory may have its own js2c variant if the "global" one does not fit its need. It seems that only the mbed and esp8266 targets use it. However, it would be more beneficial if such a tools could be evolved in a way that suits  everyone's needs, of course.)

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg