jerryscript-project/jerryscript #1076: Assertion 'ecma_get_value_type_field (value) == ECMA_TYPE_STRING' failed in ecma_get_string_from_value

renatahodovan, 2016-05-19:
###### Jerry version:
    Checked revision: e1c8a2e
    Build: debug.linux
###### OS: 
    Ubuntu 15.10, x86_64
###### Test case:
```javascript
try { Array.prototype.unshift(1)  } catch($){}
try { Object.freeze(this.Array.prototype)()  } catch($){}
try { new String(1).split("")  } catch($){}
```

###### Backtrace:
```
ICE: Assertion 'ecma_get_value_type_field (value) == ECMA_TYPE_STRING' failed at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c(ecma_get_string_from_value):548.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485b37 in syscall_2 () at jerryscript/jerry-libc/target/posix/jerry-asm.S:60
60    SYSCALL_2
(gdb) bt
# 0   0x0000000000485b37 in syscall_2 () at jerryscript/jerry-libc/target/posix/jerry-asm.S:60
# 1   0x000000000040037f in raise (sig=6) at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:92
# 2   0x0000000000400351 in abort () at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:78
# 3   0x000000000041d117 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/targets/default/jerry-port-default-fatal.c:54
# 4   0x000000000041cd58 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:68
# 5   0x000000000041cd9f in jerry_assert_fail (assertion=0x49d650 "ecma_get_value_type_field (value) == ECMA_TYPE_STRING", 
    file=0x49d468 "jerryscript/jerry-core/ecma/base/ecma-helpers-value.c", 
    function=0x48bbf0 <__func__.3353.lto_priv.489> "ecma_get_string_from_value", line=548)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:95
# 6   0x0000000000442a12 in ecma_get_string_from_value (value=16) at jerryscript/jerry-core/ecma/base/ecma-helpers-value.c:548
# 7   0x00000000004592b9 in ecma_builtin_string_prototype_object_split.lto_priv.670 (this_arg=627, arg1=858, arg2=24)
    at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-string-prototype.c:1769
# 8   0x000000000045d6b7 in ecma_builtin_string_prototype_dispatch_routine (builtin_routine_id=136, this_arg_value=627, arguments_list=0x7fffffffcb34, 
    arguments_number=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtin-string-prototype.inc.h:75
# 9   0x0000000000474aab in ecma_builtin_dispatch_routine (builtin_object_id=ECMA_BUILTIN_ID_STRING_PROTOTYPE, builtin_routine_id=136, this_arg_value=627, 
    arguments_list=0x7fffffffcb34, arguments_number=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.inc.h:55
# 10  0x000000000047442c in ecma_builtin_dispatch_call (obj_p=0x8b33c8 <mem_heap+840>, this_arg_value=627, arguments_list_p=0x7fffffffcb34, 
    arguments_list_len=1) at jerryscript/jerry-core/ecma/builtin-objects/ecma-builtins.c:531
# 11  0x000000000043a430 in ecma_op_function_call (func_obj_p=0x8b33c8 <mem_heap+840>, this_arg_value=627, arguments_list_p=0x7fffffffcb34, 
    arguments_list_len=1) at jerryscript/jerry-core/ecma/operations/ecma-function-object.c:649
# 12  0x000000000047ad99 in opfunc_call.lto_priv.799 (frame_ctx_p=0x7fffffffcba0) at jerryscript/jerry-core/vm/vm.c:379
# 13  0x000000000047a0f7 in vm_execute (frame_ctx_p=0x7fffffffcba0, arg_p=0x0, arg_list_len=0)
    at jerryscript/jerry-core/vm/vm.c:2504
# 14  0x000000000047a1bf in vm_run_with_inline_stack.lto_priv.796 (frame_ctx_p=0x7fffffffcba0, arg_p=0x0, arg_list_len=1)
    at jerryscript/jerry-core/vm/vm.c:2540
# 15  0x00000000004766d0 in vm_run (bytecode_header_p=0x8b3300 <mem_heap+640>, this_binding_value=35, lex_env_p=0x8b30b8 <mem_heap+56>, is_eval_code=false, 
    arg_list_p=0x0, arg_list_len=1) at jerryscript/jerry-core/vm/vm.c:2614
# 16  0x000000000047a9d2 in vm_run_global (error_value_p=0x7fffffffcc88) at jerryscript/jerry-core/vm/vm.c:190
# 17  0x000000000048194a in jerry_run (error_value_p=0x7fffffffce20) at jerryscript/jerry-core/jerry.c:1748
# 18  0x000000000047fbc2 in main (argc=3, argv=0x7fffffffd318) at jerryscript/main-unix.c:511
```

akosthekiss, 2016-08-29:
The issue is still valid on current master (8edab96).

LaszloLango, 2016-09-06:
#1308 is landed

closed 2016-09-06 by LaszloLango