jerryscript-project/jerryscript #1081: Assertion 'object_p->type_flags_refs >= ECMA_OBJECT_REF_ONE' failed in ecma_deref_object

renatahodovan, 2016-05-19:
###### Jerry version:
    Checked revision: e1c8a2e
    Build: debug.linux
###### OS: 
    Ubuntu 15.10, x86_64
###### Test case:
```javascript
try { new RegExp().compile(A)  } catch (err) {}
try { A } catch (err) {}
try { new RegExp().constructor.prototype.toString()  } catch (A) {}
try { A } catch (err) {}
try { new RegExp(1, "g").exec(1)  } catch (A) {}
try { A } catch (err) {}
try { A } catch (A) {}
try { A } catch (A) {}
try { new (Boolean.constructor.prototype)  } catch (err) {}
try { Boolean  } catch (A) {}
try { A } catch(A) {}
try { new RegExp().compile(new (RegExp)())  } catch (A) {}
try { Math.max().constructor.seal(A) } catch (err) {}
try { Date.parse(RegExp.prototype.compile(RegExp.prototype)) } catch(A) {}
try { Boolean.A() } catch (err) {}
```

###### Backtrace:
```
ICE: Assertion 'object_p->type_flags_refs >= ECMA_OBJECT_REF_ONE' failed at jerryscript/jerry-core/ecma/base/ecma-gc.c(ecma_deref_object):186.
Error: ERR_FAILED_INTERNAL_ASSERTION

Program received signal SIGABRT, Aborted.
0x0000000000485b37 in syscall_2 () at jerryscript/jerry-libc/target/posix/jerry-asm.S:60
60    SYSCALL_2
(gdb) bt
# 0   0x0000000000485b37 in syscall_2 () at jerryscript/jerry-libc/target/posix/jerry-asm.S:60
# 1   0x000000000040037f in raise (sig=6) at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:92
# 2   0x0000000000400351 in abort () at jerryscript/jerry-libc/target/posix/jerry-libc-target.c:78
# 3   0x000000000041d117 in jerry_port_fatal (code=ERR_FAILED_INTERNAL_ASSERTION)
    at jerryscript/targets/default/jerry-port-default-fatal.c:54
# 4   0x000000000041cd58 in jerry_fatal (code=ERR_FAILED_INTERNAL_ASSERTION) at jerryscript/jerry-core/jrt/jrt-fatals.c:68
# 5   0x000000000041cd9f in jerry_assert_fail (assertion=0x49edb0 "object_p->type_flags_refs >= ECMA_OBJECT_REF_ONE", 
    file=0x49ece0 "jerryscript/jerry-core/ecma/base/ecma-gc.c", 
    function=0x48b9f0 <__func__.3536.lto_priv.512> "ecma_deref_object", line=186)
    at jerryscript/jerry-core/jrt/jrt-fatals.c:95
# 6   0x000000000044a60d in ecma_deref_object (object_p=0x8b3b88 <mem_heap+2824>)
    at jerryscript/jerry-core/ecma/base/ecma-gc.c:186
# 7   0x000000000040e069 in ecma_lcache_invalidate_entry (entry_p=0x933810 <ecma_lcache_hash_table.lto_priv+1424>)
    at jerryscript/jerry-core/ecma/base/ecma-lcache.c:88
# 8   0x000000000040e3fc in ecma_lcache_insert (object_p=0x8b3870 <mem_heap+2032>, prop_name_p=0x8b3b88 <mem_heap+2824>, prop_p=0x0)
    at jerryscript/jerry-core/ecma/base/ecma-lcache.c:204
# 9   0x0000000000444423 in ecma_find_named_property (obj_p=0x8b3870 <mem_heap+2032>, name_p=0x8b3b88 <mem_heap+2824>)
    at jerryscript/jerry-core/ecma/base/ecma-helpers.c:727
# 10  0x0000000000444009 in ecma_create_named_data_property (object_p=0x8b3870 <mem_heap+2032>, name_p=0x8b3b88 <mem_heap+2824>, is_writable=true, 
    is_enumerable=false, is_configurable=false) at jerryscript/jerry-core/ecma/base/ecma-helpers.c:584
# 11  0x0000000000432e3d in ecma_op_create_mutable_binding (lex_env_p=0x8b3870 <mem_heap+2032>, name_p=0x8b3b88 <mem_heap+2824>, is_deletable=false)
    at jerryscript/jerry-core/ecma/operations/ecma-lex-env.c:138
# 12  0x000000000047ecc2 in vm_loop.lto_priv.800 (frame_ctx_p=0x7fffffffcba0) at jerryscript/jerry-core/vm/vm.c:2377
# 13  0x000000000047a0c7 in vm_execute (frame_ctx_p=0x7fffffffcba0, arg_p=0x0, arg_list_len=0)
    at jerryscript/jerry-core/vm/vm.c:2495
# 14  0x000000000047a1bf in vm_run_with_inline_stack.lto_priv.796 (frame_ctx_p=0x7fffffffcba0, arg_p=0x0, arg_list_len=1)
    at jerryscript/jerry-core/vm/vm.c:2540
# 15  0x00000000004766d0 in vm_run (bytecode_header_p=0x8b34b8 <mem_heap+1080>, this_binding_value=35, lex_env_p=0x8b30b8 <mem_heap+56>, is_eval_code=false, 
    arg_list_p=0x0, arg_list_len=1) at jerryscript/jerry-core/vm/vm.c:2614
# 16  0x000000000047a9d2 in vm_run_global (error_value_p=0x7fffffffcc88) at jerryscript/jerry-core/vm/vm.c:190
# 17  0x000000000048194a in jerry_run (error_value_p=0x7fffffffce20) at jerryscript/jerry-core/jerry.c:1748
# 18  0x000000000047fbc2 in main (argc=3, argv=0x7fffffffd318) at jerryscript/main-unix.c:511
```

ktorpi, 2016-07-08:
Works on current master (97be8bf).

closed 2016-07-08 by LaszloLango

corporateshill, 2017-08-18:
`eval('v_0=[[],[],[],[],[,],[,],[this,]];v_0.sort (JSON.stringify);');` also triggers

LaszloLango, 2017-08-21:
@corporateshill please open a new issue for it.