jerryscript-project/jerryscript #1086: Huge performance regression of snapshot mode

wateret, 2016-05-20:
On the [benchmark page](http://samsung.github.io/jerryscript/benchmark/benchmark.html), there is a huge performance regression recently.

This issue occurs snapshot mode only. Here are recent patches related to snapshot mode.

* e1c8a2ec96c29a8bf7fbe8a853bc3f9e18ed795b
* 7cf8b79429e3279d3de0e6f55e92a21212dc3304

cc: @LaszloLango. @zherczeg

LaszloLango, 2016-05-20:
@wateret, thanks for reporting. We'll check it.

zherczeg, 2016-05-20:
It seems there a constant 54.9 sec increase for all tests:

http://samsung.github.io/jerryscript/benchmark/details.html?date=2016-05-19&engine=jerryscript-snapshot&benchmark=sunspider-1.0.2
http://samsung.github.io/jerryscript/benchmark/details.html?date=2016-05-20&engine=jerryscript-snapshot&benchmark=sunspider-1.0.2

zherczeg, 2016-05-23:
We need some help since we could not reproduce this issue on RPi2 here. Could you send us an unstripped binary used by your benchmark system?

wateret, 2016-05-24:
@zherczeg I'm looking at my automatic benchmark script and it seems like there's some bug in it. I'll let you know when it is fixed.

wateret, 2016-05-26:
It turns out there was some problem with my benchmark script. I fixed it.

closed 2016-05-26 by wateret

zherczeg, 2016-05-26:
Thank you for investigating it.

zherczeg, 2016-05-26:
It is nice to see that we are 17% faster than Duktape. Our measurements shows similar results.