jerryscript-project/jerryscript #1092: Close dead branches

zherczeg, 2016-05-23:
Currently there are 26 open branches in Jerry. Several of them are 8 months old, and unlikely to be updated. Since we moved to form model, it would be good to only have a master branch, and perhaps each release could have one. What do you think?

akiss77, 2016-05-23:
There are certainly some dead feature branches in origin, which could and should be removed. There are some others which are somewhat questionable, however: e.g., `demo_soscon2015` may be one which we might want to keep for a longer period of time to help replicability. So, I think a case-by-case investigation is needed. But overall, sure, merged or abandoned feature & fix branches should be removed.
(And let's not forget to keep `gh-pages`...)

zherczeg, 2016-05-23:
I think the author of each branch should decide whether his/her own branches are needed. This is the first step. I have already dropped my dead branches.

lemmaa, 2016-05-24:
CC: @egavrin @sand1k @ruben-ayrapetyan @seanshpark @chunseoklee @ILyoan @wateret , Please check the branches you have created, if you have. Thank you in advance. :)

wateret, 2016-05-24:
Nothing to keep for me :D

egavrin, 2016-05-24:
Slightly reduced this number to 13  ^_^

seanshpark, 2016-05-25:
nothing for me :)

seanshpark, 2016-05-31:
Removed what I created. I think `gh-pages` should exist.

LaszloLango, 2016-07-12:
@zherczeg, I think we removed the unnecessary branches. Please check that we can close this issue or not.

zherczeg, 2016-07-13:
remotes/origin/demo_soscon2015
  remotes/origin/fix339
  remotes/origin/gh-pages
  remotes/origin/iotjs
  remotes/origin/refactor-lit-storage-rebased
  remotes/origin/snapshot-nuttx-dev
  remotes/origin/targets-rpi2

I think these branches are outdated and could be closed.

LaszloLango, 2016-07-13:
@zherczeg, we must keep 'gh-pages'

seanshpark, 2016-07-14:
In case who don't know 'gh-pages' , it's the http://samsung.github.io/jerryscript/

zherczeg, 2016-07-15:
I forgot to remove remotes/origin/gh-pages, it is needed of course.

zherczeg, 2016-07-15:
remotes/origin/fix339 by Ilyong Cho - the issue works here
remotes/origin/targets-rpi2 - @seanshpark a phony target added in 6th of january. Is it needed?
remotes/origin/snapshot-nuttx-dev - Chunseok Lee - The new API has proper strict mode support for snapshots, so I don't think this workaround is still needed.
remotes/origin/iotjs - Peter Gal - data fix - does it still needed? (patch was made in 16th of september)

seanshpark, 2016-07-18:
> remotes/origin/targets-rpi2

it is ok to remove :)

lemmaa, 2016-07-19:
cc: @ILyoan @chunseoklee

akosthekiss, 2016-08-19:
@ILyoan fix339?
@galpeter iotjs?
@chunseoklee snapshot-nuttx-dev?
(What about removing these if no answer comes back by, say, end of August?)

galpeter, 2016-08-26:
@akosthekiss @zherczeg the `iotjs` branch is not needed, we can delete it

akosthekiss, 2016-09-07:
After careful review, the snapshot-nuttx-dev branch turned out not to contain useful changes anymore, thus was deleted. As all targeted branches were removed, we can close this issue.

closed 2016-09-07 by akosthekiss