jerryscript-project/jerryscript #1094: Understanding about snapshot

jiangzidong, 2016-05-25:
Hi

My understanding about snapshot is below, please correct me if anything wrong.

- snapshot consists of 3 parts: byte-code, literal-refs and  literal-storage-data

- When call `jerry_exec_snapshot`, if we set the `copy_bytecode`, then all 3 parts will load into mem_heap, and if we unset the `copy_bytecode`, then the literal-refs and literal-storage-data will load into the mem_heap

So the snapshot's benefits are

1. for performance, save the parser time
2. for flash size, maybe the size of snapshot will samller than size of js code
3, ** for mem_heap (RAM), if we unset `copy_bytecode`, we can save the space for bytecode only. We still need mem_heap to store the literal-refs and literal-storage-data **

Thanks

Zidong

zherczeg, 2016-05-25:
Yes, you are right.

The problem with literals that they are part of the literal storage, and cannot determine where they actually are when the snapshot is loaded.

jiangzidong, 2016-05-26:
@zherczeg Thanks!

closed 2016-05-26 by jiangzidong