jerryscript-project/jerryscript #1125: Prepare the engine for release v1.0

LaszloLango, 2016-06-06:
This issue is created to collect all the tasks we need to do before the first release of the engine.

### Documentation

- [ ] GitHub wiki
- [ ] gh-pages (http://samsung.github.io/jerryscript)
- [ ] In source documentation
  - [ ] Comments in public headers files
  - [ ] Markdown files under docs folder (internals doc. and API examples&references)

### Jerry API

 - [ ] Finalize names of the API functions
 - [ ] Add missing API functions (ex.: converter functions like 'to_boolean')
 - [ ] Remove unnecessary or internal functions from public API

### Jerry Port API
 - [ ] Finalize Jerry Port API. See: #964

### Refactor the build system

 - [ ] Create new build system. See: #998

LaszloLango, 2016-06-06:
I tried to summarize the TODOs for the planned release. Anyone who has write access feel free to edit the description of the issue. Every comment and suggestion is welcome.

seanshpark, 2016-06-08:
👍

seanshpark, 2016-06-08:
Not sure how to explain but how about usage examples something like target porting how-to, small toy project examples, video clips links... ?

tilmannOSG, 2016-08-23:
I've update the issue to reflect the current status :)

polaroi8d, 2016-08-30:
Im tested the mbed OS with FRDM-K64F board and the simple blinky example. Its works right :white_check_mark:

tilmannOSG, 2016-08-30:
@polaroi8d Thanks, that's great to hear! Can you please also mention the git hash of the version that you tested?

polaroi8d, 2016-08-30:
Im tested the 'JerryScript v1.0 Release Candidate 2' source code from github.com/Samsung/jerryscript/releases site, but there is a hash too: 1e82ae1eb5b9d3da20478b660a326c58f743b586

tilmannOSG, 2016-08-30:
@polaroi8d I see, we actually merged the Curie support recently (to make sure that it will be part of the 1.0 release), those changes were not part of the RC2 tarball on Friday. Can you please do another test run with the jerryscript-release-1.0.0 branch?  
Thanks in advance!

polaroi8d, 2016-08-30:
@tilmannOSG with 7f6911103fa99c3231f234cf14fd4465b810c8d1 hash its okey?

tilmannOSG, 2016-08-30:
@polaroi8d Looks like the hash you mentioned is from trunk, please test with https://github.com/Samsung/jerryscript/tree/jerryscript-release-1.0.0 :)  
The current HEAD in jerryscript-release-1.0.0 is https://github.com/Samsung/jerryscript/commit/04e597a6a4679dc8d59dc73843a03c90d67885ed  
Can you please give it another try? :)

bzsolt, 2016-08-30:
I've checked the *x86-32 Linux*, *x86-64 Linux* and *ARMv7 Linux* targets with the precommit script (`tools/run-tests.py --precommit`) and everything works fine.
The tested hash is https://github.com/Samsung/jerryscript/commit/04e597a6a4679dc8d59dc73843a03c90d67885ed

tilmannOSG, 2016-08-30:
@bzsolt Thank you!

polaroi8d, 2016-08-31:
Sorry @tilmannOSG, now tested with the mentioned 04e597a6a4679dc8d59dc73843a03c90d67885ed hash, everything works fine with this.

tilmannOSG, 2016-08-31:
@polaroi8d No worries, thanks for doing another round of testing!

pfalcon, 2016-08-31:
Tested with a smoke test (executing `var test=0; for (t=100; t<1000; t++) test+=t; print ('Hi JS World! '+test);`) following Zephyr targets:

* qemu_x86
* qemu_cortex_cm3
* frdm_k64f (on a real device)

JrS revision used: current HEAD of jerryscript-release-1.0.0 branch, de9c27b12. Zephyr version used: v1.5.0 release (v1.5.0 ig Zephyr repo, 5ca45ba3).

Thanks!

jiangzidong, 2016-09-01:
Tested in Arduino101 with targer/CurieBSP port, everything works fine.
Jerry hash: [de9c27b](https://github.com/Samsung/jerryscript/commit/de9c27b1226188397b4eac934c598d82755e3aa0)
CurieBSP version: [V3.1.1](https://github.com/CurieBSP/main/releases/tag/v3.1.1)

tilmannOSG, 2016-09-01:
@pfalcon @jiangzidong Excellent, thanks for the test runs!

robertsipka, 2016-09-01:
I've tested the RIOT target with stm32f4 board, it works fine!
 Jerry-hash: de9c27b1
 RIOT-hash: 26e4004d

bzsolt, 2016-09-01:
https://github.com/Samsung/jerryscript/commit/de9c27b1226188397b4eac934c598d82755e3aa0 works well with NuttX on STM32F4 board.

LaszloLango, 2016-09-01:
@tilmannOSG, I've updated the GitHub wiki. Deleted 2 outdated wiki page (Roadmap and Ideas for improvement) and added references to the in source documentations. We have duplicated wiki pages: https://github.com/Samsung/jerryscript/wiki/Developer's-Certificate-of-Origin and https://github.com/Samsung/jerryscript/wiki/JerryScript-Developer's-Certificate-of-Origin are identical. I think we should keep only one of them. What do you think?

tilmannOSG, 2016-09-01:
@robertsipka @bzsolt Thanks! Looks like we have test results for all targets now :) (except for ESP8266)

tilmannOSG, 2016-09-01:
@LaszloLango Excellent, thanks for doing the cleanup. It seems the two DCO pages are identical indeed. As you say, we don't need two of them so I think you can just go ahead and delete https://github.com/Samsung/jerryscript/wiki/Developer's-Certificate-of-Origin.

LaszloLango, 2016-09-02:
JerryScript 1.0 is out, so we can close this issue.
https://github.com/Samsung/jerryscript/releases/tag/v1.0

closed 2016-09-02 by LaszloLango

jiangzidong, 2016-09-02:
great! @zherczeg Could you please modify a little bit in the description of  https://github.com/Samsung/jerryscript/releases/tag/v1.0

`Curie on Arduino101`  ==> `Curie BSP on Arduino101`

Because Curie is the SOC (HW) of Arduino101, and CurieBSP is the SDK for Curie Device :)

zherczeg, 2016-09-02:
Done

tilmannOSG, 2016-09-02:
@jiangzidong Oh I see, apologies for the mistake :)

jiangzidong, 2016-09-02:
@tilmannOSG @zherczeg Thanks!