jerryscript-project/jerryscript #1148: Jerry API v1.0

LaszloLango, 2016-06-15:
### Planned API changes

 - [x] Change 'jerry_api' prefix to 'jerry_'
 - [x] Remove 'jerry_value_t' structure from API
 - [ ] Unify error handling
 - [ ] Remove 'jerry_error_t' (duplicate of 'ecma_standard_error_t')
 - [ ] Add missing API functions
   - [ ] Converter functions (ToBoolean, ToNumber, ToObject)
   - [ ] Array helpers (ex.: GetLength, GetElements)
 - [ ] Remove unnecessary or internal functions from public API

grgustaf, 2016-06-15:
I think the jerry_value_is_object/string/etc. functions could be inlined?
Also, take a look at jerry_string_to_char_buffer. I think it should work like strncpy, but instead last I looked I think it puts a '\0' at the end of the given buffer, regardless of how many characters were copied, so there may be uninitialized data left between the end of the string and the end of the buffer. In other words, put a '\0' at the end of the string, if it fits within the buffer.

zherczeg, 2016-06-16:
I would prefer them to be inlined myself, but we are talking about a C API where inlining is not a trivial task. You need all-in-one or LTO build to exploit inlining. We could make them macros. Or do you have a better suggestion?

LaszloLango, 2016-06-16:
@grgustaf, the jerry_strint_t is not a zero terminated string. It can contain '\0' anywhere, so currently there is no '\0' insertion to the end. There is a note for this in the source file:
```
 * Note:
 *   '\0' could occur in characters.
```

At this moment it is caller's responsibility to initialize the buffer if it can be bigger than the string. Only the size defines the end of the string. Do you need an API function to get zero terminated strings from `jerry_string_t`?

LaszloLango, 2016-06-16:
I've updated the description with new tasks.

grgustaf, 2016-06-18:
I was coming in to ask for set_data/get_data (native handle) but it looks like you added that recently. That will be huge for me, thanks. :)

Regarding strings, what is the encoding then? Not UTF-8 I guess? Sorry that I haven't dug in to find out myself. Not sure if I need another function, depending on the answer. What I've needed so far is a way to compare JS strings against known constant strings from C; that's what I'm intending when I use jerry_string_to_char_buffer.

LaszloLango, 2016-06-20:
@grgustaf, the engine uses CESU-8 encoding for strings.

closed 2016-07-20 by LaszloLango