jerryscript-project/jerryscript #1180: I can't download the stm32f3discovery_fw.zip

nisen, 2016-07-01:
$ make prerequisites
```
Prerequisites setup failed. See /home/fmj/osokay/jerryscript/build/prerequisites/prerequisites.log for details.
Makefile:396: recipe for target 'prerequisites' failed
make: *** [prerequisites] Error 1
```

$ cat ./build/prerequisites/prerequisites.log 
```
Setting up stm32f3 prerequisite
--2016-07-01 20:29:16--  http://www.st.com/st-web-ui/static/active/en/st_prod_software_internet/resource/technical/software/firmware/stm32f3discovery_fw.zip
Resolving www.st.com (www.st.com)... 114.80.90.218, 114.80.90.216
Connecting to www.st.com (www.st.com)|114.80.90.218|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://www.st.com/web/en/st_prod_software_internet/resource/technical/software/firmware/stm32f3discovery_fw.zip [following]
--2016-07-01 20:29:16--  http://www.st.com/web/en/st_prod_software_internet/resource/technical/software/firmware/stm32f3discovery_fw.zip
Reusing existing connection to www.st.com:80.
HTTP request sent, awaiting response... 404 Not Found
2016-07-01 20:29:16 ERROR 404: Not Found.

Failed to setup 'stm32f3' prerequisite. Cannot download 'http://www.st.com/st-web-ui/static/active/en/st_prod_software_internet/resource/technical/software/firmware/stm32f3discovery_fw.zip' zip archive.
```

i use browser download the url is error too. 

'''
curl 
'''

nisen, 2016-07-01:
the error is not same as https://github.com/Samsung/jerryscript/issues/598

tilmannOSG, 2016-07-01:
@nisen It seems like ST Microelectronics is no longer hosting the respective files under those links.  
By looking at http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32-standard-peripheral-libraries-expansions/stsw-stm32068.html it appears that they now require you to click through a license agreement and register on their website before you can download anything. This obviously breaks the current model of installing the dependencies automatically.  
We probably need to remove this completely and just ask the users/developers to install the board firmware package manually.  
@zherczeg @LaszloLango Can you please double-check on the ST Microelectronics web site and see whether you can still find the files there?

tilmannOSG, 2016-07-01:
Adding @poussa I believe he ran into the same issue a couple of days ago.

zherczeg, 2016-07-04:
We are currently reworking the build system of Jerry, and we plan to remove this dependency. Unfortunately ST changed the availability of this file and the archive must be downloaded manually after accepting a license agreement.

LaszloLango, 2016-07-12:
Related PR: #1192

akosthekiss, 2016-08-19:
The build system has been reworked, and `make prerequisites` together with ST bare metal example app and automated ST dependencies installation are gone. Thus, closing this issue. (If you think it needs further discussion, feel free to re-open it. Also, if you find other issues with the project, we are happy to have them reported.)

closed 2016-08-19 by akosthekiss