jerryscript-project/jerryscript #1184: jerry --exec-snapshot date-tostring fails to execute

tilmannOSG, 2016-07-04:
In the Travis CI ARM build of https://github.com/Samsung/jerryscript/pull/1182, the following test failed:  
[37/266] build/bin/debug.linux/jerry  --exec-snapshot date-tostring.t9Fkk0uDHJ.snapshot: FAIL (1)
Script error: assertion failed  
The full log is available at https://travis-ci.org/Samsung/jerryscript/jobs/142165136.  
It's pretty much impossible that https://github.com/Samsung/jerryscript/pull/1182 actually caused this failure since it only changed a Zephyr-specific Makefile and a README, e.g. no code changes.  
This is either a recently introduced non-deterministic regression or an issue with our test harness.

zherczeg, 2016-07-04:
Actually this flaky test is there for a long time, and improssible to reproduce the crash. Even if it is reported by travis, and the binary is downloaded, the crash does not happen.

tilmannOSG, 2016-07-04:
I see, thanks for the clarification. Definitely sounds like something non-deterministic. Might be worth to run the test with AddressSanitizer/MemorySanitzer/Valgrind to get to the bottom of this. I would guess it's probably an uninitialized variable or a pointer in a hash table causing this non-determinism.  
Did this show up on x86-64 as well? Or is it specific to ARM?

zherczeg, 2016-07-04:
We spent some time to figure this out (with valgrind), but no success at all. Might have some dependency which is not available in our environment. It is a date test after all.

akosthekiss, 2016-08-19:
What to do with this issue? Close with "who knows why?" or keep it open as a reminder (but still, who knows why?...)

LaszloLango, 2016-08-20:
IMHO, keep open for a while.

zherczeg, 2016-08-20:
We could not reproduce it on other machines so far. Very difficult to debug.

tilmannOSG, 2016-09-06:
@zherczeg Mentioned to me that this issues has not triggered a single time after the build system rewrite. Could very well have been a bug in the test harness of the old build system. Closing for now, feel free to reopen in case the issue shows up again.

tilmannOSG, 2016-09-13:
Reopening, managed to trigger the bug again: https://travis-ci.org/Samsung/jerryscript/jobs/159634805  
Should be possible to reproduce with a507c08.

LaszloLango, 2018-02-16:
It looks like this was fixed. Feel free to reopen if needed.

closed 2018-02-16 by LaszloLango