jerryscript-project/jerryscript #1188: main-unix.c on Ubuntu: jerry_create_external_function assert not reachable (reference error)

zippocage, 2016-07-06:
I have been trying to add external function handlers but without success. I saw that there is already an external function declared in main-unix.c. When i write a small test script to check that the assert function works it also gives me "uncaughtException: ReferenceError" (same as when I try to add an external function myself).

I am building and testing this on Ubuntu 14.04 and jerryscript is built and run through iotjs as a dependence.

Am I doing something wrong or is there some issues with the jerry_create_external_function?

zherczeg, 2016-07-06:
Could you paste here your source code?

zippocage, 2016-07-06:
The test script:
`assert(false)`

and the external function that is declared in main-unix.c is actually already in the checked in code:
```
  jerry_object_t *global_obj_p = jerry_get_global ();
  jerry_object_t *assert_func_p = jerry_create_external_function (assert_handler);
  jerry_value_t assert_value = jerry_create_object_value (assert_func_p);

  bool is_assert_added = jerry_set_object_field_value (global_obj_p,
                                                       (jerry_char_t *) "assert",
                                                       assert_value);

  jerry_release_value (assert_value);
  jerry_release_object (global_obj_p);

  if (!is_assert_added)
  {
    jerry_port_errormsg ("Warning: failed to register 'assert' method.");
  }

```

Maybe the issue is that the external function (here assert) is not defined as a magic string and that is the reason that the reason for reference error? I tried to just change the "assert" to "my_assert" above to avoid possible collision with existing assert function but that did not help.

zherczeg, 2016-07-06:
This should work if it placed in the right place (after jerry_init()). Maybe the global object is replaced by another global somehow.

Btw a new API is under construction at the moment. See #1177.

zippocage, 2016-07-06:
Going through the code I only see one call to jerry_init (for the default target) and the code right after is the code posted above (row 316-330 in main-unix.c). There is also no other calls to jerry_set_object_field_value that could affect the global object that I can find.

What happens if you try to call assert in a script or maybe you are not running on Ubuntu?

galpeter, 2016-07-06:
@zippocage Do I understand correctly that you are running your script with iotjs (that is you are not using the jerry binary for your test js file)? If so then the assert is registered there differently.

zippocage, 2016-07-06:
@galpeter correct, I am using the iotjs binary to run the script. So jerry_init and jerry_run etc are executed in another way when used by iotjs? If so, it would explain why this is not working. Is the use of external function not possible in conjunction with iotjs perhaps or just have to be careful when/where it is done?

closed 2016-07-06 by zippocage

galpeter, 2016-07-06:
@zippocage iotjs does not use the main-unix.c file, it has it's own entry point. So you need to put your code which registers the extern functions there.