jerryscript-project/jerryscript #1218: JerryScript web site broken

tilmannOSG, 2016-07-20:
Now that the jerryscript.net domain has been enabled the web site is working again. However there seem to be some path issues with the web site e.g. if you access it through jerryscript.net the browser can't find the CSS style sheet anymore.  
Visiting http://jerryscript.net should show exactly the same web site as is shown when visiting through http://samsung.github.io/jerryscript.

tilmannOSG, 2016-07-20:
Adding @ktorpi as he probably knows how to fix this :)

LaszloLango, 2016-07-20:
@tilmannOSG, my guess is that this is what you need: https://github.com/Samsung/jerryscript/tree/gh-pages/css

ktorpi, 2016-07-20:
Is this http://jerryscript.net site also generated by github?

tilmannOSG, 2016-07-20:
@ktorpi Yes, it is the same web site (there's only one). GitHub is redirecting http://jerryscript.net to http://samsung.github.io/jerryscript. From what I can see in the HTML sources we have some links to .../jerryscript which now break I guess.

tilmannOSG, 2016-07-26:
#1226 landed yesterday but it seems the change hasn't propagated to the GitHub CDN yet.

tilmannOSG, 2016-07-27:
We had a longer discussion about this on IRC today, turns out that the three recent changes to the gh-pages branch are not showing up on http://jerryscript.net. The oldest change is now 8 days old. There seems to b problem in getting the web site changes to the GitHub CDN (which serves jerryscript.net), the changes show up just fine on http://samsung.github.io/jerryscript.  
I've contacted GitHub support to see whether there's something they can do about that.

tilmannOSG, 2016-07-28:
Ok after speaking with GitHub support I managed to fix this. Apparently for some reason the jerryscript.net custom domain was pointing to a forked repository of JerryScript. GitHub support cleared the custom domain from that fork now.  
Also for project web sites you have to add the CNAME file to the gh-pages branch rather than the master branch, I wish the GitHub help pages would state that explicitly :)  
In retrospective this all makes sense. After we updated the DNS records I was surprised that jerryscript.net started working immediately (showing the broken page) without me having to change any GitHub settings. I assume this happened because there was some fork out there which had a CNAME file in the repository :)  
Closing this issue as the web site works just fine now.

closed 2016-07-28 by tilmannOSG