jerryscript-project/jerryscript #1220: Nominating Dániel Bátyai (dbatyai) for JerryScript Maintainer status

zherczeg, 2016-07-20:
Hi,

I would like to nominate Akos Kiss to become a JerryScript maintainer. He already have reviewed 73 pull requests, and landed 56 patches. The list of these follows.

Reviews:
#1208 Remove arbitrary case conversion tests. 
#1204 Improve resolve reference.
#1203 Minor fixes in API reference
#1201 Implement IO port API
#1200 Improve the construction of "length" built-in strings.
#1197 Update API reference to the new API
#1195 More gc-friendly property hashmap allocation.
#1192 WIP: Re-thinking the build system to bring it more into line with the conv…
#1189 Simplify string management.
#1186 Small performance optimizations of the interpreter.
#1183 Optimize comparison in VM.
#1177 Jerry API update
#1173 Optimize arithmetic in VM.
#1170 Sixteen bit hash for strings
#1169 Fix memcpy return value.
#1168 Optimize literal storage.
#1167 Check error values in API functions
#1166 Documentation is moved to docs.
#1165 Four byte optimized memcpy.
#1164 Remove ECMA_OBJECT_TYPE_BUILT_IN_FUNCTION to free an object type.
#1163 Create extended objects instead of internal properties. 
#1161 Add new API function
#1160 Update API reference
#1157 Update API examples
#1154 Remove LIST_LAZY_PROPERTY_NAMES_ROUTINE_NAME macro function.
#1150 Add conversion API functions
#1149 Improve builtin-property construction.
#1147 Update jerry API
#1142 Remove string copy
#1140 Rename ecma_is_value_error to ECMA_IS_VALUE_ERROR. 
#1138 Use 'ecma_value_t' in Jerry API
#1137 Rename try_give_memory_back to a better free_unused_memory.
#1136 Improve get-value
#1135 Optimize put result phase in the interpreter.
#1132 Parser improvements.
#1127 Move branches into a separate interpreter group.
#1126 Optimize add operation.
#1124 Update API functions
#1122 Remove last_completion_value.
#1121 Reduce the default LCache size by half. From 4Kbyte to 2Kbyte.
#1120 Improve equal operations.
#1119 Modify the usage of ecma_string_to_utf8_string()
#1116 Optimize LCache operation.
#1113 Improve empty checks of ecma string
#1107 Optimize uint32 hash computation.
#1105 Remove ECMA_STRING_CONTAINER_HEAP_NUMBER.
#1104 Optimize conditional jumps.
#1101 Code simplification
#1100 Replace multiple boolean arguments to one uint8_t.
#1093 Fix value passing in JerryScript
#1039 Introduce integer ecma-value representation to reduce the double allocations
#1034 Add context support for memory allocator
#976 Remove MEM_HEAP_PTR_64 macro
#635 Remove ECMA_STRING_CONTAINER_CONCATENATION type from ecma_string
#633 Check setTime call's this_arg
#631 Fix tr arguments in the Makefile bug infrastructure
#629 Check if there is really an old bytecode when recompiling a regexp
#606 Global object unescape routine  development ecma builtins normal
#569 Fix undefined initialValue handling in Array.prototype.reduce and reduceRight function
#562 Fix the this argument passed to string replace callbacks.
#543 Use ecma-collection in JSON.
#512 Fix regexp match on right brackets
#467 Make parseInt unicode tolerant
#455 Implement Date.prototype.toJSON()
#452 Improve Infinity handling in Math.min/max methods
#443 Implement String.prototype.match
#430 Fix parseInt when passing empty string
#315 Fix ecma_op_array_object_define_own_property
#199 Implement Date object helper functions
#124 Implement Object.keys and Object.getOwnPropertyNames
#99 Implement Object.prototype.propertyIsEnumerable().
#96 Implement Array.prototoype.toLocaleString()
#86 Implement Array.prototype.splice().
#38 Implement Array.prototype.join

He also did major contributions to JerryScript such as creating a memory allocator or CESU-8 string optimizations:

Pull requests:
#1207 Optimize ecma_string_get_array_index
#1024 Add some asserts to check heap structure consistency
#1004 Print unhandled errors in REPL mode
#1000 Add help flag and usage to UNIX Jerry standalone.
#999 Fix syscall assertion when input file does not exist 
#874 New Allocator and improved String handling.
#678 Use heap allocated buffer instead of char collection in ecma_strings
#616 Change internal encoding of strings to CESU-8
#579 Implement AnnexB RegExp.prototype.compile()
#568 Fix undefined fromIndex in Array.prototype.lastIndexOf()
#567 Parse integer numbers from back to front.
#554 Fix RegExp literal parsing
#552 Fix invalid control escapes during RegExp parsing.
#547 Fix bound function length
#539 Trim whitespace from argument names in Function constructor
#527 Fix assertion fails in parseInt()
#518 Fix array index normalize in case index is negative zero
#493 Use ToNumber when checking lastIndex during RegExp exec()
#492 Fix RegExp constructor when called with undefined arguments
#481 Handle Unicode characters in parseFloat()
#462 Fix push() and unshift() in case result length is larger than UINT_MAX
#458 Use [[DefineOwnProperty]] in Array builtins where necessary.
#457 Fix Array index normalize helper when index is large.
#450 Fix Array.prototype.toLocaleString() when toLocaleString returns non-string.
#449 Fix buffer size calculation in Number.prototype.toString()
#441 Fix Number.prototype.toString() when radix is undefined.
#434 Returned array of Array.prototype.map() should have same length as original.
#427 Fix leak in Array.prototype.indexOf()
#415 Fix Array.prototype.concat() when 'this' is not an array.
#400 Fix Array.prototype.concat() argument array length.
#382 Number.prototype functions should throw TypeError if 'this' is not number.
#335 Array.prototype.sort() should sort in place.
#326 Add length property to Array constructor
#325 Fix [[Class]] property of builtin constructors
#306 Implement printing in the specified radix for Number.prototype.toString().
#298 Fix Number.prototype.toFixed if argument is outside int32 range 
#238 Implement Number.prototype.toPrecision()
#225 Fix buffer size in Number.prototype.toFixed()
#224 Implement Number.prototype.toExponential()
#207 Implement printing in the specified radix for Number.prototype.toString().
#190 Implement parseFloat()
#167 Implement parseInt()
#98 Implemented Array.prototype.concat()
#95 Implemented Array.prototype.sort()
#90 Implemented Array.prototype.reverse()
#89 Implemented Array.prototype.filter()
#85 Implemented Array.prototype.some() 
#84 Implemented Array.prototype.every()
#77 Implemented Array.prototype.lastIndexOf(). 
#39 Implemented Array.prototype.forEach(). 
#26 Implemented Array.prototype.indexOf(). 
#25 Implemented Array.prototype.unshift(). 
#24 Array.prototype.shift(). 
#23 Implemented [[HasProperty]] for Objects. 
#22 Implemented Array.prototype.pop().

(Note: this is a retroactive nomination)

lemmaa, 2016-07-20:
Why do we put off until now? Strongly in favor. :)

akosthekiss, 2016-07-20:
I second the nomination

zherczeg, 2016-08-01:
There was no objection so Daniel is a maintainer.

closed 2016-08-01 by zherczeg