jerryscript-project/jerryscript #1221: Native functions cannot be named "catch"

jprestwo, 2016-07-21:
When implementing Promises, I found that a JS script will not run at all if there is a native function named "catch". For promises, this is needed in order to by syntactically correct e.g.

mypromise.then(function(ret) {
 . . . 
    }).**catch**(function(error) {
. . . 
});

zherczeg, 2016-07-21:
What do you mean by "not run"?

    print("A");
    mypromise.then(function(ret) {
    }).catch(function(error) {
    });

This prints A and then throws a ReferenceError, which is expected.

jprestwo, 2016-07-21:
When I create an object, add a function property to it named "catch", call that function, and run the JavaScript, it does nothing. No error or anything. This was also with a lot of other variables in play so I will write some simplified code and post it.

zherczeg, 2016-07-21:
var promise = {};
    promise.catch = function() {
      print("CATCH");
    };
    promise.catch()

Prints CATCH, so it works in JS.

jprestwo, 2016-07-21:
Sorry about this, false alarm. Our project runs our JS samples through a minifier and it was silently NOT writing the JS correctly into the final string (when it detected that catch was not being used correctly i.e. not in a try catch block) that gets compiled! This can be closed.

zherczeg, 2016-07-21:
No problem.

closed 2016-07-21 by zherczeg