jerryscript-project/jerryscript #1223: Nominating Somang Park for JerryScript Steering Committee Member

lemmaa, 2016-07-22:
Hi,

I would like to nominate Somang Park(@somang-park) to become a member of [Steering Committee](https://github.com/Samsung/jerryscript/wiki/Governance#steering-committee), as the main role of Community Strategy Director. She already work for JerryScript project as a community maintainer in Samsung Open Source Group and willing to actively expand her work to make JerryScript be a one of successful public open source project in the world.

Here attaches her letters for joining steering committee.

Evgeny Gavrin(@egavrin), Zoltan Herczeg(@zherczeg) and all the contributors of JerryScript, I'm asking your opinion respectfully.
Thank you in advance.

——
Letters from Somang Park

>Hi, I'm responsible for discovering the new open source projects from Samsung. Also I'm fully involved in launching number of new open source projects and supporting open source communities including PR for projects. 
>Now I'm concentrating on the Jerryscript.
> 
>I've been encouraging the participant company's activity for JerryScript so that it can be further growth as a representative technology of JavaScript in IoT and Wearables finally. For this, I've discussed with the companies which are already interested in JerryScript such as Intel, Pebble, ARM, and have plan putting it in Open Source Foundation (Now, It's jQuery Foundation but, might be changed to cover not only jQuery but also whole JavaScript technology) to include a large number of developers and echo.
> 
>Currently, JerryScript is under the mentorship program of jQuery Foundation. I'm willing to make JerryScript more neutral project. So after graduating the mentorship program and satisfying the condition for re-launching the project in jQuery Foundation, JerryScipt will move out of "github/samsung".
>
>From very first step of this plan, I've working on with Min-Jung Kim the former community maintainers of JerryScript. I've made plan and shared with other companies, also prepared to be participate in the foundation, supported the mentorship program and all process. I'll manage overall contents or issues which can be occurred during the mentorship program such as IP, Infra, Governance including cooperation with partners in future as a steering committee member.
So community of users and contributors through evangelism and marketing efforts grows and gets stability by ensuring continuity through supporting efforts.
>
>I expect my role makes easier to operate project with a number of member companies, be accepted our(JerryScript side) requests in community, gather members, and arrange effective PR in representative open source events such as Node.js Interactive, OSCON or ApacheCon.
> 
>Thank you,
>Somang Park
>

egavrin, 2016-07-22:
+1
I support the nomination and believe that this will help us greatly expand engine's adoption.

lemmaa, 2016-07-29:
Based on your comment, I updated the [maintainer list](https://github.com/Samsung/jerryscript/wiki/Assigned-people) on the wiki.
Somang, welcome to join us! :)

closed 2016-07-29 by lemmaa