jerryscript-project/jerryscript #1224: Question: Ways to decrease the initial heap consumption?

slaff, 2016-07-24:
I am trying to use JerryScript on ESP8266 device.  This device has around 40K free. My app, without JerryScript compiled in it, has around 35032 bytes free heap. Just by adding JerryScript with CONFIG_MEM_HEAP_AREA_SIZE = 10K my free heap decreased to 8976 bytes. I tried to use all compact options and decrease the  CONFIG_MEM_HEAP_AREA_SIZE to 4K and ended up with 14680 bytes free heap. Decreasing CONFIG_MEM_HEAP_AREA_SIZE further down did not bring any improvement. 
Which means that minimum initial heap usage from JerryScript for me was 35032 - 14680 = 20352 bytes. 

My question is - are there other options to decrease the initial heap usage from JerryScript? Why JerryScript is reserving so much initial memory? Is it possible to decrease the initial heap size and expand it when it is needed?

zherczeg, 2016-07-25:
There is a static property cache, which consumes 2K memory and can be disabled by CONFIG_ECMA_LCACHE_DISABLE. Also a there are some globals around 100 bytes. So in your configuration Jerry consumes around 12K. I have not idea about the rest of the memory consumption, perhaps your compiler does not support const arrays, and load them into the RAM instead of just keeping them in the ROM.

tilmannOSG, 2016-07-25:
@slaff When running on an ARM Thumb-2 device our base footprint of JerryScript is at around ~3.5KB of RAM (when executing JerryScript on an empty .js file). I know the ESP8266 uses an Xtensa CPU but I would assume the memory consumption should be quite similar, did you get a chance to measure the base memory footprint with an empty .js file yet?

slaff, 2016-07-26:
> There is a static property cache, which consumes 2K memory and can be disabled by CONFIG_ECMA_LCACHE_DISABLE

@zherczeg Thanks for the hint. I will try it later and report if it worked for me.

> on an ARM Thumb-2 device our base footprint of JerryScript is at around ~3.5KB of RAM (when executing JerryScript on an empty .js file). 

@tilmannOSG I checked the heap before loading a file and after loading a file and there was negligible difference in the heap usage. 

I am compiling JerryScript core only and using the math library from microc/gcc. Is it recommended to use JerryScript's math (jerry-libm)?

Also I am compiling JerryScript core with the following Makefile. Can it be that I am doing something wrong here?

```Makefile
-include Makefile.local

TOOLCHAIN_PREFIX := xtensa-lx106-elf-
CC := $(TOOLCHAIN_PREFIX)gcc
AR := $(TOOLCHAIN_PREFIX)ar
LD := $(TOOLCHAIN_PREFIX)gcc
AR := $(TOOLCHAIN_PREFIX)ar
OBJCOPY := $(TOOLCHAIN_PREFIX)objcopy

XTENSA_LIBS ?= $(shell $(CC) -print-sysroot)
ifeq ($(XTENSA_LIBS),)
	XTENSA_LIBS = $(ESP_HOME)/xtensa-lx106-elf/
endif

BUILD_BASE	= out/build

SRC_DIR := jerry-core \
			jerry-core/ecma/base \
			jerry-core/ecma/builtin-objects \
			jerry-core/ecma/operations \
			jerry-core/jmem \
			jerry-core/jrt \
			jerry-core/lit \
			jerry-core/parser/js \
			jerry-core/parser/regexp \
			jerry-core/vm

SRC			:= $(foreach sdir,$(SRC_DIR),$(wildcard $(sdir)/*.c*))
BUILD_DIR	:= $(addprefix $(BUILD_BASE)/,$(SRC_DIR))
OBJ_FILES	:= $(patsubst %.c,%.o,$(SRC))

INCDIR		:= $(addprefix -I,$(SRC_DIR))

LDFLAGS  += -L$(XTENSA_LIBS)/lib \
			-L$(XTENSA_LIBS)/arch/lib \

CFLAGS+=-std=c99 -D__TARGET_ESP8266

# CFLAGS += -Wall -Os -g -O2 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mno-text-section-literals  -D__ets__ -DICACHE_FLASH

CFLAGS += -fno-builtin -fno-stack-protector -g -gdwarf-4 -Wall -Werror=all -Wextra -Werror=extra -Wformat-nonliteral -Werror=format-nonliteral -Winit-self -Werror=init-self -Wconversion -Werror=conversion -Wsign-conversion -Werror=sign-conversion -Wformat-security -Werror=format-security -Wmissing-declarations -Werror=missing-declarations -Wno-stack-protector -Wno-attributes -Wlogical-op -Werror=logical-op  -D__attr_always_inline___= -Wl,-EL -fno-inline-functions -ffunction-sections -fdata-sections -mlongcalls -mtext-section-literals -mno-serialize-volatile -pedantic -Os
#ifeq ($(COMPACT),"1")
	CFLAGS += -DCONFIG_ECMA_COMPACT_PROFILE
#endif 

CFLAGS += -DCONFIG_ECMA_NUMBER_TYPE=CONFIG_ECMA_NUMBER_FLOAT32 -DCONFIG_MEM_HEAP_AREA_SIZE=4096 \
		  -DJERRY_ENABLE_SNAPSHOT_EXEC -DJERRY_ENABLE_SNAPSHOT_SAVE -DJERRY_NDEBUG -D_BSD_SOURCE \
		  -D__TARGET_ESP8266 -D__attr_always_inline___=

CFLAGS += $(INCDIR) $(SDK_INCDIR)

BIN_DIR := bin
AXTLS_AR := $(BIN_DIR)/libjerrycore.a

V ?= $(VERBOSE)
ifeq ("$(V)","1")
Q :=
vecho := @true
else
Q := @
vecho := @echo
endif

all: patchjmp $(AXTLS_AR) $(BUILD_DIR)
rebuild: clean all

$(BUILD_DIR):
	$(Q) mkdir -p $@

$(AXTLS_AR): | $(BIN_DIR)

$(AXTLS_AR): $(OBJ_FILES)
	for file in $(OBJ_FILES); do \
		$(OBJCOPY) \
		--rename-section .text=.irom0.text \
		--rename-section .literal=.irom0.literal \
		$$file; \
	done
	$(AR) cru $@ $^
	
patchjmp: $(AXTLS_AR)
	$(Q) $(AR) x ${XTENSA_TOOLS_ROOT}/../xtensa-lx106-elf/sysroot/lib/libcirom.a lib_a-setjmp.o
	$(Q) $(AR) r ${AXTLS_AR} lib_a-setjmp.o
	$(Q) rm lib_a-setjmp.o

$(BIN_DIR):
	$(Q) mkdir -p $(BIN_DIR)

clean:
	$(Q) rm -rf $(OBJ_FILES) $(AXTLS_AR) $(BUILD_DIR)

.PHONY: all clean
```

tilmannOSG, 2016-07-26:
@slaff Looking closer at the ESP8266 hardware I think what's happening is that some constant data ends up in the data memory (which could be offloaded to flash memory on an STM32F4 board). My understanding is that the flash in the ESP8266 only allows for aligned 32-bit reads and that's why the linker puts it in data memory instead.  
Can you please run "readelf" on your binary so we can see where the different sections end up?

zherczeg, 2016-07-26:
Jerry has a large amount of const data (mainly strings which cannot be read as 32 bit words) which is usually put into the NOR flash section on ARM platforms, so it can be accessed without using RAM (similar to instructions). So no RAM is wasted on such data.

I checked some discussions in forums, and it seems your compiler puts const data into RAM by default, and not easy to force it to do otherwise. This is a big surprise for me, I thought all low-memory platforms try to optimize read-only data.

slaff, 2016-07-26:
@tilmannOSG readelf with output of all data looks like this

```
ELF Header:
  Magic:   7f 45 4c 46 01 01 01 00 00 00 00 00 00 00 00 00 
  Class:                             ELF32
  Data:                              2's complement, little endian
  Version:                           1 (current)
  OS/ABI:                            UNIX - System V
  ABI Version:                       0
  Type:                              EXEC (Executable file)
  Machine:                           Tensilica Xtensa Processor
  Version:                           0x1
  Entry point address:               0x40100004
  Start of program headers:          52 (bytes into file)
  Start of section headers:          3072780 (bytes into file)
  Flags:                             0x300
  Size of this header:               52 (bytes)
  Size of program headers:           32 (bytes)
  Number of program headers:         5
  Size of section headers:           40 (bytes)
  Number of section headers:         248
  Section header string table index: 245

Symbol table '.symtab' contains 3538 entries:
   Num:    Value  Size Type    Bind   Vis      Ndx Name
     0: 00000000     0 NOTYPE  LOCAL  DEFAULT  UND 
     1: 3ffe8000     0 SECTION LOCAL  DEFAULT    1 
     2: 3ffe8390     0 SECTION LOCAL  DEFAULT    2 
     3: 3ffef028     0 SECTION LOCAL  DEFAULT    3 
     4: 40202010     0 SECTION LOCAL  DEFAULT    4 
     5: 40100000     0 SECTION LOCAL  DEFAULT    5 
     6: 00000000     0 SECTION LOCAL  DEFAULT    6 
     7: 00000000     0 SECTION LOCAL  DEFAULT    7 
     8: 00000000     0 SECTION LOCAL  DEFAULT    8 
     9: 00000000     0 SECTION LOCAL  DEFAULT    9 
    10: 00000000     0 SECTION LOCAL  DEFAULT   10 
    11: 00000000     0 SECTION LOCAL  DEFAULT   11 
    12: 00000000     0 SECTION LOCAL  DEFAULT   12 
    13: 00000000     0 SECTION LOCAL  DEFAULT   13 
    14: 00000000     0 SECTION LOCAL  DEFAULT   14 
    15: 00000000     0 SECTION LOCAL  DEFAULT   15 
    16: 00000000     0 SECTION LOCAL  DEFAULT   16 
    17: 00000000     0 SECTION LOCAL  DEFAULT   17 
    18: 00000000     0 SECTION LOCAL  DEFAULT   18 
    19: 00000000     0 SECTION LOCAL  DEFAULT   19 
    20: 00000000     0 SECTION LOCAL  DEFAULT   20 
    21: 00000000     0 SECTION LOCAL  DEFAULT   21 
    22: 00000000     0 SECTION LOCAL  DEFAULT   22 
    23: 00000000     0 SECTION LOCAL  DEFAULT   23 
    24: 00000000     0 SECTION LOCAL  DEFAULT   24 
    25: 00000000     0 SECTION LOCAL  DEFAULT   25 
    26: 00000000     0 SECTION LOCAL  DEFAULT   26 
    27: 00000000     0 SECTION LOCAL  DEFAULT   27 
    28: 00000000     0 SECTION LOCAL  DEFAULT   28 
    29: 00000000     0 SECTION LOCAL  DEFAULT   29 
    30: 00000000     0 SECTION LOCAL  DEFAULT   30 
    31: 00000000     0 SECTION LOCAL  DEFAULT   31 
    32: 00000000     0 SECTION LOCAL  DEFAULT   32 
    33: 00000000     0 SECTION LOCAL  DEFAULT   33 
    34: 00000000     0 SECTION LOCAL  DEFAULT   34 
    35: 00000000     0 SECTION LOCAL  DEFAULT   35 
    36: 00000000     0 SECTION LOCAL  DEFAULT   36 
    37: 00000000     0 SECTION LOCAL  DEFAULT   37 
    38: 00000000     0 SECTION LOCAL  DEFAULT   38 
    39: 00000000     0 SECTION LOCAL  DEFAULT   39 
    40: 00000000     0 SECTION LOCAL  DEFAULT   40 
    41: 00000000     0 SECTION LOCAL  DEFAULT   41 
    42: 00000000     0 SECTION LOCAL  DEFAULT   42 
    43: 00000000     0 SECTION LOCAL  DEFAULT   43 
[... 2804 lines cut ...]
  2848: 40212130   187 FUNC    GLOBAL DEFAULT    4 wifi_fpm_set_sleep_type
  2849: 4024aafc    33 FUNC    GLOBAL DEFAULT    4 _Z8fileSeeksi15SeekOrigin
  2850: 40257f90   247 FUNC    GLOBAL DEFAULT    4 ecma_find_or_create_liter
  2851: 40205e10    18 FUNC    GLOBAL DEFAULT    4 wifi_rfid_locp_recv_close
  2852: 4024574c     7 FUNC    WEAK   DEFAULT    4 _ZN15IDelegateCallerIvJcP
  2853: 4025d230   153 FUNC    GLOBAL DEFAULT    4 parser_stack_pop
  2854: 40105e74    77 FUNC    GLOBAL DEFAULT    5 _ZN5Timer12initializeMsEj
  2855: 3fff4764     4 OBJECT  GLOBAL DEFAULT    3 g_cnx_probe_rc_list_cb
  2856: 402151e0    69 FUNC    GLOBAL DEFAULT    4 wDev_Crypto_Conf
  2857: 40258c28    20 FUNC    GLOBAL DEFAULT    4 ecma_builtin_type_error_t
  2858: 40212e0c    51 FUNC    GLOBAL DEFAULT    4 reset_noise_timer
  2859: 4022210c    85 FUNC    GLOBAL DEFAULT    4 espconn_recv_unhold
  2860: 40223d90   101 FUNC    GLOBAL DEFAULT    4 netif_set_ipaddr
  2861: 40246b20   207 FUNC    WEAK   DEFAULT    4 _ZN7HashMapIc6StringE8all
  2862: 3ffef028     0 NOTYPE  GLOBAL DEFAULT  ABS _rodata_end
  2863: 402232ac   119 FUNC    GLOBAL DEFAULT    4 ethernet_input
  2864: 40203a54     8 FUNC    GLOBAL DEFAULT    4 system_init_done_cb
  2865: 4024bbb8    89 FUNC    GLOBAL DEFAULT    4 _ZN10MqttClient17onReadyT
  2866: 4024a3a8    27 FUNC    GLOBAL DEFAULT    4 _ZNK6String5toIntEv
  2867: 402082e4   100 FUNC    GLOBAL DEFAULT    4 ram_pbus_xpd_tx_on
  2868: 40203ccc     8 FUNC    GLOBAL DEFAULT    4 wifi_station_dhcpc_status
  2869: 40228e30    27 FUNC    GLOBAL DEFAULT    4 espconn_igmp_leave
  2870: 3ffede2c    48 OBJECT  GLOBAL DEFAULT    2 ecma_builtin_object_proto
  2871: 4023d4d4   400 FUNC    GLOBAL DEFAULT    4 asn1_get_private_key
  2872: 40253b80   279 FUNC    GLOBAL DEFAULT    4 sp

zherczeg, 2016-07-26:
It seems all OBJECT starts with 3xx address, and everything else (FUNC, NOTYPE, ...) starts with 4xx. 

    a small const array (8*4=32 bytes): ecma_property_hashmap_steps
    451: 3ffebb30    32 OBJECT  LOCAL  DEFAULT    2 ecma_property_hashmap_ste

Do all OBJECT goes to RAM?

zherczeg, 2016-07-26:
https://github.com/esp8266/esp8266-wiki/wiki/Memory-Map

3FFE8000h-3FFFC000h 	dram0

Yes OBJECTs seem to be mapped there.

slaff, 2016-07-26:
> I checked some discussions in forums, and it seems your compiler puts const data into RAM by default, and not easy to force it to do otherwise. 

@zherczeg It looks like you are right. I found the following related discussions: 
https://github.com/SuperHouse/esp-open-rtos/issues/161
https://github.com/espruino/Espruino/issues/697
https://github.com/SuperHouse/esp-open-rtos/issues/11
http://www.esp8266.com/viewtopic.php?f=9&t=6376&start=1

slaff, 2016-07-26:
I looked at the examples code in the Espressif SDK and it seems that I can achieve this by adding two more attributes.

For example if I have a constant declared like this:

```
const char flash_str[] = fmt;
```

I can add the following two attributes
```
const char flash_str[] ICACHE_RODATA_ATTR STORE_ATTR = fmt;
```

Where ICACHE_RODATA_ATTR and  STORE_ATTR are defined as follows
```
#define ICACHE_RODATA_ATTR  __attribute__((section(".irom.text")))
#define STORE_ATTR __attribute__((aligned(4))) 
```

@zherczeg @tilmannOSG 
So my question is - is there an easy way to add these two attributes to all constants OR is JerryScript having some preprocessor directive that I can overwrite?

zherczeg, 2016-07-27:
I don't see any easy way to add these to all constants. But you can track them down using the readelf and search OBJECT type (this is specific to your architecture). You can also propose a patch which adds a macro to all const data (empty string by default). I don't see a reason why we couldn't land it.

slaff, 2016-07-27:
> You can also propose a patch which adds a macro to all const data (empty string by default). I don't see a reason why we couldn't land it.

@zherczeg Sounds good to me. I will try to make a Pull Request in the coming week(s).

tilmannOSG, 2016-07-27:
@slaff Obviously we can't really sacrifice code readability by wrapping every string in a macro just for the sake of one platform but I'm sure we have several files where a lot of constants are defined in one place and for those it should be relatively easy to move them over to flash.  
Based on the numbers you posted it sounds like the JerryScript constants eat up 40% (16KB) of the available ~40KB of RAM for user applications, would be really nice if we could get that number down :)

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg