jerryscript-project/jerryscript #1247: Add jerry_... APIs to create NaN and inf

martijnthe, 2016-08-05:
Would it make sense to have wrappers for `ecma_make_nan_value()` and `ecma_number_make_infinity()` ?

zherczeg, 2016-08-05:
Do you mean public API wrappers? Technically it is possible. However I somehow feel that these functions would be better suited for math libraries, since you can convert doubles to jerry numbers.

It seems gnu math.h has them:
https://www.gnu.org/software/libc/manual/html_node/Infinity-and-NaN.html

martijnthe, 2016-08-05:
Yeah, I mean public API wrappers indeed.
I couldn't find an easy way to create a `NaN` value.

zherczeg, 2016-08-08:
Please check PR #1252

martijnthe, 2016-08-08:
Looks good, thanks much!

LaszloLango, 2016-08-14:
IMHO, We should consider to add jerry_value_is_infinite and jerry_value_is_nan as well.

akosthekiss, 2016-08-19:
The PR that added the two requested functions has been landed some days ago. Shall we close this issue or shall we keep it open as a reminder that two more API functions could be added, as suggested by @LaszloLango ? (Anyone for or against those two funcs?)

zherczeg, 2016-08-20:
I think I will close it, and can be reopened if more work is done.

closed 2016-08-20 by zherczeg