jerryscript-project/jerryscript #1248: jerryscript.net links all give 404 errors

hbp, 2016-08-06:
The web site jerryscript.net has bad links giving 404 errors. For example the Internals link send you to http://jerryscript.net/jerryscript/internals/ instead of  http://jerryscript.net/internals/ ditto for all other links on the home page.

cookiengineer, 2016-08-09:
How can I help you guys to fix the `gh-pages` branch? It somehow seems to miss many contents, like the API examples. Is the `gh-pages` branch the real server that hosts the jerryscript.net domain?

Would love to help you guys with the website, so you can focus more on the runtime.

cookiengineer, 2016-08-09:
I created a pull request (already linked) that fixes the base url messup. The links afterwards should work, the new domain jerryscript.net is used (as already present in the CNAME file).

tilmannOSG, 2016-09-12:
#1340 Brings the web site into a working state again.  
Now tools/update-webpage.sh needs to be updated accordingly (to emit site.github.url) because otherwise the next manual sync of the web site will lead to a broken page again.

LaszloLango, 2016-11-16:
Fixed.

closed 2016-11-16 by LaszloLango