jerryscript-project/jerryscript #1262: [Q & Suggestion] Does jerryscript run generational GC?

esevan, 2016-08-12:
After I analyze the memory management code of jerryscript, I've got a question about WHITE_GRAY GC pool and BLACK GC pool.
 According to the code, the objects in WHITE_GRAY pool are gonna be swept after visiting process and ones in BLACK pool remain in memory. However, once gc process done, BLACK pool is converted to WHITE_GRAY pool as below in order to check them at later GC.
```
/* Unmarking all objects */
  ecma_object_t *black_objects = JERRY_CONTEXT (ecma_gc_objects_lists) [ECMA_GC_COLOR_BLACK];
  JERRY_CONTEXT (ecma_gc_objects_lists)[ECMA_GC_COLOR_WHITE_GRAY] = black_objects;
  JERRY_CONTEXT (ecma_gc_objects_lists) [ECMA_GC_COLOR_BLACK] = NULL;
```
Question is here. It seems insufficient to regard current jerryscript GC as a generational GC, even though it runs GC in SEVERITY_LOW condition only if newly allocated objects are more than threshold and it only checks WHTE_GRAY pool in GC process because there's no object in BLACK pool at start of GC.

What if the objects in BLACK pool is not moved to WHITE_GRAY pool after each GC so as not to check older objects in the memory at later GC? I think it makes the GC more generational, despite of no assurance about perf and memory usage before check. :)

If I understood the GC process wrongly, please fix me.

zherczeg, 2016-08-12:
Jerry gc is just mark and sweep. It could be improved though.

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg