jerryscript-project/jerryscript #1264: [esp8266] build failure

seanshpark, 2016-08-13:
While applying new build system, I get this error when compiling `esp8266/user` folder.
```
make[1]: Entering directory '/home/maxwell/usbdisk/x86/harmony/jerryscript/targets/esp8266/user'
Makefile:42: *** Too many open files.  Stop.
make[1]: Leaving directory '/home/maxwell/usbdisk/x86/harmony/jerryscript/targets/esp8266/user'
/opt/Espressif/ESP8266_SDK/Makefile:317: recipe for target '.subdirs' failed
make: *** [.subdirs] Error 2
```

commit https://github.com/Samsung/jerryscript/commit/23db8b5544133cc5791e513300eb7bfd2adb1d3b#diff-93f1dba5fbae34b11a322c6bd3269549 
seems the reason.
```
 INCLUDES := $(INCLUDES) -I $(PDIR)include
 -INCLUDES += -I ./ -I ../include
 -PDIR := ../$(PDIR)
 +INCLUDES += -I ./ -I ../include -I../../../
  sinclude $(PDIR)Makefile
```

`PDIR := ../$(PDIR)` line should not be removed. 

I'll make a PR fixing this error and for the new build system.

seanshpark, 2016-08-15:
@slaff, is there any reason to include `../../..` folder?

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg