jerryscript-project/jerryscript #1270: Number as a 32-bit float

grgustaf, 2016-08-16:
I suppose this must already be well-understood, but I left a test with a counter running over the weekend, and it got stuck at 16777216. Doing 16777216 + 1 returns 16777216. I discovered this is due to the resolution limitation of the 32-bit float.

I see in the JerryScript internals docs that this is an intentional decision, following one of the two formats from IEEE 754. However, when I look here it seems to suggest that ECMAScript is supposed to use the 64-bit format.

http://www.ecma-international.org/ecma-262/5.1/#sec-4.3.19

It sure seems like 16 million is pretty low for math to stop working. Is this seen as acceptable and a limitation JerryScript programmers should be aware of?

zherczeg, 2016-08-17:
This was exactly my motivation when the I recommended the change the default from 32 bit float to 64 bit float. Most people are not aware of the implications of float 32 limitations (e,.g. limits binary logical operations), so using the float 64 is better as default. However, if somebody is aware of the limitations (and his device only has 32 bit float acceleration), he can still switch to float.

zherczeg, 2016-08-17:
Please check #1229

akosthekiss, 2016-08-20:
Given that #1229 has been landed several weeks ago and 64 bit floats are the default, is this still a valid issue (or can we close it)?

tilmannOSG, 2016-08-23:
Yes, this issue can be closed. With #1229 the default behavior of JerryScript is ECMAScript 5.1 compliant again and the option to use 32-bit floats only remains for users who are willing to trade precision against performance and who are aware of the implications of doing that.  
@grgustaf Feel free to reopen if you still see any issues on trunk :)

closed 2016-08-23 by tilmannOSG