jerryscript-project/jerryscript #1274: Large string truncated without error

svaarala, 2016-08-18:
In trying to do some performance testing I noted that when trying to build a large string it gets silently truncated to zero bytes:

```
var txt = 'xy';
for (var i = 0; i < 20; i++) {
    print(i, txt.length);
    txt = txt + txt;
}
print(txt.length);
```

This should print:

```
0 2
1 4
2 8
3 16
4 32
5 64
6 128
7 256
8 512
9 1024
10 2048
11 4096
12 8192
13 16384
14 32768
15 65536
16 131072
17 262144
18 524288
19 1048576
2097152
2097152
```

But prints:

```
0 2
1 4
2 8
3 16
4 32
5 64
6 128
7 256
8 512
9 1024
10 2048
11 4096
12 8192
13 16384
14 32768
15 0
16 0
17 0
18 0
19 0
0
```

This is probably caused by an internal limitation on string length? What was confusing and is a bit dangerous is that the result is silently truncated rather than triggering a `RangeError`.

What seems to happen is an overflow in the length field because the result is not always zero length. For example:
```
x = 'x'; while (x.length < 65530) x += 'y';
print(x.length);
x += 'dummydummy';
print(x.length);
print(x);
```

This prints:
```
65530
4
xyyy
```

So that the contents are from the beginning of the original string before appending.

Assuming this is a simple 16-bit overflow, the string append processing should check for overflow and cause a RangeError if there would be an overflow. Without the error the behavior is a bit dangerous; in performance testing it caused the test to run but with incorrect inputs, for instance.

zherczeg, 2016-08-18:
Thank you for the report, I will fix this in due course. There was only an assert for this.

zherczeg, 2016-08-23:
Since 1275 is accepted, we can close this bug.

closed 2016-08-23 by zherczeg

svaarala, 2016-08-23:
Just to be clear, the 16-bit string length limitation is quite understandable for low memory environments, but the silent length wrapping is dangerous because a script can proceed silently with incorrect data. Typically engines throw e.g. a `RangeError` when encountering internal limits and I think that would be appropriate here.