jerryscript-project/jerryscript #1295: Unused build ID defines

akosthekiss, 2016-08-29:
The CMakeLists of jerry-core calculates and defines `JERRY_BUILD_DATE`, `JERRY_COMMIT_HASH`, and `JERRY_BRANCH_NAME`. I recall having some const strings in the past with these values which allowed the identification of the build during runtime, but they are not used in the code anymore anywhere. Instead, we have `JERRY_API_*_VERSION` defined in jerry-api.h (`jerry -v` prints "Version: 1.0" now).

Shall we keep the build ID macros in the build system and re-introduce the variables (e.g., into jerry-api.h and jerry.c), or are they not needed anymore? (In which case the related logic could be removed from CMakeLists.)

LaszloLango, 2016-08-30:
IMHO, they are not needed anymore.

tilmannOSG, 2016-08-30:
I think we can drop JERRY_BUILD_DATE (build becomes non-deterministic) and JERRY_BRANCH_NAME (do we really need that?) but it would be nice to keep JERRY_COMMIT_HASH and have a way to print it to the console when running the binary. This makes it easier to identify which sources the binary originated from.

akosthekiss, 2016-08-30:
I agree, build date and branch name don't contain any useful information.

Regarding the commit hash, some questions (so that it's not happening _after_ a PR gets opened):
* Should we put it in the library or in the command line tool? The latter is easier, but the former provides all users of the library with a buid ID at runtime. However, that would need some API extension (access to a const string, or a function that returns such a string; we cannot put that into a macro like `JERRY_API_MAJOR_VERSION`, at least not with the current build system, as the public API header is not configurable, and the hash ID comes from the build system).
* Should we use full hash ID or the short one (7 chars)?

tilmannOSG, 2016-09-07:
@akosthekiss Command line tool and short hash sound great to me :)

commit 714e8d26 (2016-09-07): Add git commit hash to jerry version info

Previously, jerry-core was built with various build identification
info (build date, git commit hash, git branch name), which was both
available from string constants of the library and also exposed by
jerry-main via the `--version` command line option. As of late,
jerry-core identifies itself only with API version number macros,
but sometimes it could be helpful to get access to the git commit
hash as well, even if that info does not become part of the API.

This patch moves the git commit hash retrieval logic from the build
scripts of jerry-core to  [... cut]
closed 2016-09-07 by akosthekiss