jerryscript-project/jerryscript #1296: Disallow failures of Travis CI jobs on OS X?

akosthekiss, 2016-08-29:
Currently, OS X Travis CI jobs may fail without turning the build status red. However, the landing of #1269 made OS X jobs pass. The question is whether we should "promote" OS X jobs to voting status?

Cons:
  - OS X is not on the list of official development environments.

Pros:
  - It can give a better test coverage. (E.g., the issue fixed by #1269 was core the memory subsystem, still, no test environment revealed it but OS X.)
  - If the OS X jobs remain in the "allow failures" category, the current good status may degrade easily. (It is easy to focus on the general green/red status indicator at PRs and not check all CI jobs one-by-one.)

LaszloLango, 2016-08-30:
I don't have a Mac, so I cannot debug any problem with that platform. So I'd keep OS X jobs in the allow failures category. Who will maintain the platform?

tilmannOSG, 2016-08-30:
I would be quite in favor of promoting macOS to become a fully supported platform but I agree that we need an official maintainer for that.  
@akosthekiss how did you fix the macOS issues? Travis CI obviously has support for macOS builds but I guess using that for debugging purposes is quite awkward?

akosthekiss, 2016-08-30:
@tilmannOSG I started using a mac lately, but I admit that that's a constrained resource both hardware and human-wise. (Perviously, I used Travis to track down some issues -- by uploading patches to own branch in own repo and looking at the logs --but that's a somewhat inconvenient method.)

tilmannOSG, 2016-08-30:
@akosthekiss Would you be willing to become the official maintainer of the macOS port of JerryScript? :)

akosthekiss, 2016-08-30:
@tilmannOSG As there are not much OS-specific things in either the library or in the build system, I expect few issues that need OSX maintenance. Thus, I might give the maintainership a try and hope that it can be done even with my limited resources.

tilmannOSG, 2016-08-31:
@akosthekiss Thanks Akos!  
I guess we can close this issue :)

closed 2016-08-31 by tilmannOSG