jerryscript-project/jerryscript #1297: Change name of public API header?

akosthekiss, 2016-08-29:
The 1.0 release is approaching, but before it happens, we might reconsider the name of the public API header of the JerryScript library. Now, it is called jerry-api.h, which is somewhat unusual. For an ordinary library user, writing `#include <jerry.h>` might be more intuitive.

The time of the release is close and the change might need adaptation at various places of the project, but we should think about it since whatever we choose will most probably stay with us for good (or, for a long time, at least), as the name of the header is also part of the fixed API.

(Note: I'm not proposing to change the name of jerry-port.h. That's OK, IMHO, as it is important for and used by port developers only.)

zherczeg, 2016-08-30:
I don't mind to rename the header. I think the question is how risky this change is.

LaszloLango, 2016-08-30:
I think it is not the right time to change the header name. Such change implies an RC3. It will delay the release if we rename it.

tilmannOSG, 2016-08-30:
In terms of release I think it's too late for an invasive change like that, we should only cherry-pick bug fixes for critical bugs which would block the release for one reason or another :)  
In terms of header naming I think "jerryscript.h" would be even more appropriate since after all the project is called JerryScript not Jerry :)

akosthekiss, 2016-08-30:
OK, let's wait with renamings then. (BTW: calling the resulting library libjerry-core.a is also a bit unnatural, libjerry.a or libjerryscript.a would seem more clean.)

Removing the issue from the 1.0 release milestone but leaving it open as a reminder that we might still consider something like this in the future. (Feel free to close it if you disagree, though.)

tilmannOSG, 2016-08-30:
Yes, I'd very much in favor of changing the header name to "jerryscript.h" and the library name to "libjerryscript.a" in the future :)  
I think we should consider this for the next stable release after 1.0.

closed 2017-03-04 by zherczeg