jerryscript-project/jerryscript #1305: Implement String.prototype.charCodeAt()

kenchris, 2016-08-30:
```The charCodeAt() method returns an integer between 0 and 65535 representing the UTF-16 code unit at the given index (the UTF-16 code unit matches the Unicode code point for code points representable in a single UTF-16 code unit, but might also be the first code unit of a surrogate pair for code points not representable in a single UTF-16 code unit, e.g. Unicode code points > 0x10000). If you want the entire code point value, use codePointAt().```

https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/String/charCodeAt

zherczeg, 2016-08-31:
This function is supported by Jerry. Is there any problem with it?

kenchris, 2016-08-31:
Hey @brianjjones you said this wasn't supported, but apparently it should be

akosthekiss, 2016-08-31:
With latest jerry, I get:
```
jerry> "Hi Kenneth!".charCodeAt(7)
101
jerry> "\u1234".charCodeAt(0)
4660
```

I think we are good. Can we close the issue?

kenchris, 2016-08-31:
Yeah, let's close it. I will talk to Brian :)

closed 2016-08-31 by kenchris

brianjjones, 2016-08-31:
Odd, don't know what to say it didn't work for me.  I'll try again and report back

brianjjones, 2016-08-31:
Strange, it works fine from me in jerryscript directly.  Perhaps I had a typeo, I'll let you know if I have any further issues, sorry for the noise.

brianjjones, 2016-09-01:
Ah it was fromCharCode that isn't working for me not charCodeAt.

zherczeg, 2016-09-01:
In theory that should work as well:

    String.fromCharCode(65, 66, 67);

I tried this and produced the expected "ABC"