jerryscript-project/jerryscript #1319: how can i share a c variable to jerrry readonly

dbskccc, 2016-09-05:
suppose I have a huge struct to export to javascript, but i don't want javascript to modify the value.

seems now every value have to make a copy to vm, two copy in C and jerrry is a waste memory and it's hard to sync every value changed to jerry.

zherczeg, 2016-09-05:
I would recommend to create an external function and it returns the requested value. A string argument could represent the name of the value (since you have a big struct).

dbskccc, 2016-09-06:
now, I change every single value to a function like below:
struct Ttest{
double balance;
char* name;
int history[10];
}
property balance become to an external function obj.balance(), in jerry read by call balance() and write by call obj.balance(newblance);

but for array property history[10], it's still no good way to operate the value.

zherczeg, 2016-09-06:
I think you just need to pass an index argument to the function, i.e. obj.getHistory(4); obj.setHistory(5, "newvalue") or just checking the number of parameters (1 - read, 2 - write)

dbskccc, 2016-09-06:
yeah, got it.
for array we can make it a object property and access like below:
read: obj.history.at(0) obj.history.length()
write:obj.hostory.at(1,newValue)

closed 2016-09-06 by dbskccc