jerryscript-project/jerryscript #1354: APACHE 2.0 makes forking and integration a little hard to contribute and integrate into OSS

eljefedelrodeodeljefe, 2016-09-16:
Is it by chance planned to move to a MIT license? APACHE 2.0 is sort of a blocker for thinking about contributing or using it.

akosthekiss, 2016-09-16:
Could you please elaborate on why Apache 2.0 would be a blocker for you? Which clauses cause problems? Where do you have conflicts?

tilmannOSG, 2016-09-20:
@eljefedelrodeodeljefe I'd be curious to hear about potential issues as well. Note that for software intended to be released under a liberal license the Free Software Foundation explicitly recommends to use the Apache License 2.0 instead of BSD, MIT, etc. (https://www.gnu.org/licenses/license-list.en.html#apache2).

kryptine, 2016-09-28:
The Apache License is incompatible with GPLv2. This can become a problem when combining with GPLv2-only software (such as Linux)

akosthekiss, 2016-09-28:
@kryptine AFAIK most software released under GPLv2 has the option of using GPLv3 terms, and Apache 2.0 is deemed GPLv3-compatible even by FSF. And the interpretation of ASF also allows the inclusion of Apache 2.0-licensed code in GPLv3 software (http://www.apache.org/licenses/GPL-compatibility.html).

(BTW, Linux as the kernel is just an example, right? Although it is an example of a GPLv2-and-not-any-later-version software. But one wouldn't want to include JerryScript in the Linux kernel, I guess...)

Is stepping up to GPLv3 not an option for you?

zherczeg, 2016-09-29:
LLVM also wants to move to Apache 2.0 license:
https://www.phoronix.com/scan.php?page=news_item&px=LLVM-Considers-Apache-2.0

Today software patents are a great pain for the industry, and there must be some defense against them.

kryptine, 2016-10-01:
@zherczeg After a developer summit and nearly a year it hasn't happened. I wonder what happened to the plan

kryptine, 2016-10-01:
Facebook uses a BSD 3-Clause copyright license with a separate patent license, so Apache 2.0 is not necessary to solve patent problems. https://github.com/facebook/react Also BSD/MIT license with separate patent license is smaller than Apache License 2.0

akosthekiss, 2016-10-01:
There are several examples of all the mentioned open source licences being used "out there" by big and small projects and companies, be it MIT, BSD, Apache, GPL, and variants. (And also of licenses that haven't been mentioned.) As far as I'm concerned, most of them can work together (with GPLv2 being a bit problematic in terms of compatibility but the later version option solves the problem -- and the original issue description is mentioning MIT anyway).

I may be wrong of course. So, @eljefedelrodeodeljefe & @kryptine, could you please name the projects that want to use JerryScript but cannot because of license conflicts? Or name the clauses of the current license that are blocking you personally from contributing? Or those that are conflicting with your company policies perhaps? Let's see whether we can help solving the issues.

kryptine, 2016-10-02:
Apache License, section 4:
[...]
(b) You must cause any modified files to carry prominent notices stating that You changed the files; and
(c) You must retain, in the Source form of any Derivative Works that You distribute, all copyright, patent, trademark, and attribution notices from the Source form of the Work, excluding those notices that do not pertain to any part of the Derivative Works
[...]

4b is unreasonable, why do I have to tag every file I have changed? Just an overall notice that the code has been changed is enough.
4c requires me to reproduce "all copyright, **patent, trademark** and attribution notices" (emphasis mine). I do not want to repeat the "X is a registered trademark of Y" stuff. It feels like advertising for another company.

Though, the real issue is that I don't like GPL3 due to the extra restrictions it adds, for example tivoization. Since you already use a non-copyleft license, I don't think those restrictions are valuable for you

zherczeg, 2016-10-02:
Well, if I understand correctly those sections protects you and especially your users. If the software grants a patent, and you remove it, your users can be sued. Law has lots of inconveniences, but in the end they provide a way to protect your own rights.

tilmannOSG, 2016-10-03:
@eljefedelrodeodeljefe & @kryptine The Apache License 2.0 is an established court-tested liberal license with good protection against patent issues. In that sense it provides a stable environment both for JerryScript contributors and users.  
I understand that the conditions in section 4 can cause some extra work when integrating JerryScript into a project/product and in theory we could add a similar exception as is proposed at http://lists.llvm.org/pipermail/llvm-dev/2016-September/104778.html for the LLVM project. This would allow you to skip all the attribution stuff. But then again if you consider how much work went into JerryScript from the various contributors it seems to be fair to at least ask for some attribution when the project is being used as part of other projects/products.  
In terms of GPLv2 compatibility it seems you'll get a different answer depending on which lawyer you ask. Certainly no one has actually tested this in court.
As @akosthekiss has mentioned already, we are primarily interested in understanding whether there are specific issues in the community where someone wants to integrate JerryScript with a GPLv2-only project but is worried about the legal consequences of doing so.

akosthekiss, 2016-11-01:
OP seems to have abandoned the issue and even active participants haven't continued the thread for a month now. Closing the issue, but feel free to reopen it if you feel that questions have been left unanswered.

closed 2016-11-01 by akosthekiss

AlexMax, 2017-02-27:
@tilmannOSG I have been looking for a scripting language to use with a game engine that is GPLv2 without a "or later" clause.  The use of Apache License 2.0 means that I cannot consider this JavaScript implementation, which is a shame because this project looks pretty active.

I appreciate the reasons why the Apache license was chosen - I was going to suggest the LLVM GPLv2 exception, but it looks like you are already aware of it.

Would adding such an exception be onerous?  If so, no big deal, there are other JS implementations out there.