jerryscript-project/jerryscript #1355: Consider JavaScript tools instead of python

thegecko, 2016-09-16:
Is there any interest in swapping/adding tools to be written in JavaScript (node.js) instead of python?

The reason for this would be to remove the dependency on python for building. We've found that moving to a framework for developing with JavaScript pretty much guarantees the user has node installed, so moving to that for tools would remove the requirement to install python and friends for tooling.

I'm happy to swap the tools we have for JavaScript equivalents, but won't bother if it's of little interest.

zherczeg, 2016-09-18:
I hope those who wrote those scripts will comment, but for me anything is good. Especially if you are willing to do the job :) I have thought python is installed by every distribution, or is there an exception?

akosthekiss, 2016-09-18:
My two pence: to me, this would be just swapping one dependency for another. I'd see the beauty in using a _JerryScript_-based tool set (in the spirit of "eat your own dog food" concept), but I'm not sure we are there yet. (Note: I'm a bit biased towards python because of personal background, I admit.)

thegecko, 2016-09-18:
Python certainly isn't available on every system. Windows? My observation is that there needs to be a dependency and I argue users of JerryScript are more likely to already be node users, too.

galpeter, 2016-09-18:
I've did a quick check on what node.js (npm) uses when building c/c++ packages. What I've seen is that in the background you need to have node-gyp installed which is basically a wrapper around gyp (and gyp is written in python). On the node-gyp github page it is even mentioned that you need python for it. So afaik if we would switch to node.js/npm build scripts we would also need python. (Please note this is really a quick glance on the npm packaging).

thegecko, 2016-09-18:
If that is the case, it completely nullifies my suggestion :)

zherczeg, 2016-09-19:
Can we close this issue?

LaszloLango, 2016-09-19:
My personal preference is to keep python for tools. IMHO it is quite easy to install and use on various operating systems.

thegecko, 2016-09-19:
Closing as interest seems negligible

closed 2016-09-19 by thegecko