jerryscript-project/jerryscript #1367: "Number.prototype.toFixed()" is not properly implemented

bsdelf, 2016-09-20:
The current implementation fails with following test case:

* `assert((0.995).toFixed(2) === "0.99");`
* `assert((9.995).toFixed(2) === "9.99");`

Reference:
* http://www.ecma-international.org/ecma-262/5.1/#sec-15.7.4.5

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

LaszloLango, 2017-01-31:
The bug is still exists. Please don't close this issue.

robertsipka, 2017-03-16:
When I look at the ECMAScript 5.1 specification for .toFixed() (section 15.7.4.5), I do not see it explicitly describe a solution for this kind of rounding problem here, although I see the root of the problem. Floating points are represented in binary, not decimal, so they are not compatible with basic math on decimal numbers. Therefore, a lot of JS engine gives the following results:

> (7.995).toFixed(2) == "8.00"   // "8.00" with jerryscript
> (8.995).toFixed(2) == "8.99"   // "9.00" with jerryscript

In Jerryscript, we use basic math on decimal numbers to determine the outcome of the `Number.prototype.toFixed()` method, which maybe gives more accurate results, but it is different from the expected. Probably to get closer to the usual/expected, but maybe less appropriate solution, we need a few modification in the related code parts or completely rewrite this section.  

Do you see anything what I have not noticed? What are your opinions? We should keep the current implementation or try to change it?

PR #1924 (2017-08-09): Improve toFixed function
Fixes #1367.
From now numbers are represented as binary floating-point which guarantees the expected operation of toFixed function.

**Benchmark result** 

Benchmark | RSS (bytes) | Perf (sec) |
----: | ----: | ----: | 
3d-cube.js | 57344 -> 57344 : 0.000% | 0.868 -> 0.863 : +0.615% | 
3d-raytrace.js | 126976 -> 120539 : +5.069% | 1.060 -> 1.049 : +0.994% | 
access-binary-trees.js | 53248 -> 53248 : 0.000% | 0.561 -> 0.565 : -0.573% | 
access-fannkuch.js | 20480 -> 20480 : 0.000% | 2.192 -> 2.184 : +0.327% | 
access-nbody.js | 28672 -> 28672 : 0.000% | 1.105 -> 1.102 : [... cut]
commit ad608e30 (2017-08-09): Improve toFixed function

Fixes #1367.
From now numbers are represented as binary floating-point which guarantees the expected operation of toFixed function.

JerryScript-DCO-1.0-Signed-off-by: Robert Fancsik frobert@inf.u-szeged.hu
closed 2017-08-09 by yichoi