jerryscript-project/jerryscript #1370: Heap requirement for Array.pop() gets very high for large arrays (proportional to the array's length)

martijnthe, 2016-09-21:
When calling `.pop()` on a large array, an unsurprisingly large amount of heap is required to complete the call.

It seems like this call-chain ends up being the called more than`array.length` times, each time allocating 8 bytes of heap:
- `jmem_pools_alloc` <= 8 bytes alloc'd for every 
- `ecma_alloc_collection_chunk`
- `ecma_append_to_values_collection`
- `ecma_op_object_get_property_names`
- `ecma_op_array_object_define_own_property`
- `ecma_op_object_define_own_property`
- ecma_op_general_object_put
- ecma_op_object_put
- ecma_builtin_array_prototype_helper_set_length
- ecma_builtin_array_prototype_object_pop
- ...

See also https://github.com/Samsung/jerryscript/blob/master/jerry-core/ecma/operations/ecma-array-object.c#L288

I also noted a local `uint32_t` array gets created that is sized to the number of property names (`array.length` + a couple more).

Would it be possible to reduce the required memory by avoiding the creation of the property names collection and additional local array? Any hints on doing so? Thanks.

martijnthe, 2016-09-21:
On a similar note, the `in` operator implementation also calls `ecma_op_object_get_property_names` and therefore also has a heap footprint proportional to the number of properties. Would it be possible to implement this by iterating over the existing property list instead? It would be a little tricky, because the properties could be mutated by the `for (... in ...) {}` code block, but it's not impossible I think?

Should I file a separate issue for this one?

zherczeg, 2016-09-21:
The problem with for-in is that you can delete and insert new items, and the for-in must honor delete and ignore insert. This makes things difficult.

zherczeg, 2016-09-21:
At this moment I am partially working on reworking the property management of jerryscript. A new implementation for setting an array length is on my list. It has some challenges. However, if you wish to work on this I can help you.

zherczeg, 2016-09-21:
var o = { x:5, y:6 };
    for (var i in o) {
      print(i);
      delete o.y;
      o.z = 7;
    }

The output is 'x'. Do you have a good suggestion how to do this without collecting the property names first?

martijnthe, 2016-09-21:
> Do you have a good suggestion how to do this without collecting the property names first?

Maybe the iterator could somehow maintain a list of the newly inserted objects? I guess it would be pretty gnarly/ugly, because [[Put]] would have to check whether the object is currently being iterated  (and there can be nested loops...) and if so, add it to list of each of the iterators? Hrm.. yeah, pretty hairy...

zherczeg, 2016-09-22:
Yes, an object can participate in multiple for-in statements, and these for-in statements can use different versions of the object (with different prototype chains). So we would need to track changes for each object in the prototype chain. A bit complicated task. However, if we could detect the simple (and most frequent) case somehow, that could help.

zherczeg, 2016-09-22:
Another interesting example:

    var o = { x:1, y:2, z:3 }; Object.setPrototypeOf(o, { y:2 });
    for (var i in o) { print(i); delete o.y; }

The output is "x y z"
If I delete y:2, the output is "x z y"

So in the first case we need to remember that o had an y property, even though it was deleted at some point.

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

closed 2017-01-31 by zherczeg