jerryscript-project/jerryscript #1375: ESP8266 target revival

akosthekiss, 2016-09-23:
The ESP8266 target has been in a bad shape for a while. I've tried bringing it up to speed again and got to a point where it's building and linking. However, I have the handicap of not owning a real device. So, instead of opening a pull request, I open this issue and point to the branch that contains the current status (https://github.com/akosthekiss/jerryscript/tree/esp-revival) in the hope that someone can take it over from here.

Notes:
* I've used the integrated SDK for ESP8266 from https://github.com/pfalcon/esp-open-sdk, which seems to be a NOOS SDK, unlike the previous approach which seemed to have FreeRTOS dependencies.
* I'm really "flying blindfolded", following my best developer instincts, but I have no idea whether the rewritten code has any chance to work.
* The linked binary contains too much code for default linker script, so I had to use one that allowed more space for the irom region.
* The build system stops after the ELF file is generated, so no automated flashing supported for now.
* There are still FIXMEs in the code...

CC: @seanshpark @pfalcon @slaff ?

slaff, 2016-09-23:
@akosthekiss I will take a look later today and let you know.

akosthekiss, 2016-10-01:
@slaff any success?

slaff, 2016-10-17:
@akosthekiss I tried it but it was constantly throwing fatal errors on my device :(

```
3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x40209a64, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
```

LaszloLango, 2016-11-28:
**Status update**: I rebased the branch of @akosthekiss to the current master and made some fixes. You can find it here: https://github.com/LaszloLango/jerryscript/tree/esp-revival
It is still not working properly, but should not crash:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 816, room 16 
tail 0
chksum 0x8d
load 0x3ffe8000, len 788, room 8 
tail 12
chksum 0xcf
ho 0 tail 12 room 4
load 0x3ffe8314, len 288, room 12 
tail 4
chksum 0xcf
csum 0xcf

2nd boot version : 1.2
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size : 32Mbit
jump to run usererror user bin flag, flag = c
user code done
```

@slaff, the reason of your error was a missing LD script that must be included.
`INCLUDE "eagle.rom.addr.v6.ld"`

LaszloLango, 2016-11-30:
@slaff, @pfalcon any idea what's wrong with the `user_init`? I think `jump to run usererror user bin flag, flag = c` in the output means something wrong with the user app. I tried to put an inifinte loop into my user_init, but the boot log says `user code done`, which is strange. It looks like the `user_init` never called.

pfalcon, 2016-12-03:
@LaszloLango : Sorry for the delay with responding to this ticket. To provide some background: my contributions to JerryScript are done as part of my dayjob. It would be conflict of interest to work on ESP8266 related matters at my work time. In my free time, I'm a maintainer of https://github.com/pfalcon/esp-open-sdk ,  but it has a very specific goal: when ESP8266 initially appeared, it was a collection of leaked documentation and toolchain; my interest was to de-proprietaritize it as much as possible. In that regard, esp-open-sdk is a Makefile which allows to build Xtensa toolchain from source and pulls in as few proprietary components as possible. There're still a lot of proprietary, undocumented things, and I don't know them all.

With the above in mind, ESP8266 has the following bootloading process: there's a 1st-stage bootloader in chip's (mask) ROM. It loads a code from external FlashROM and runs it. In the simplest case, this "code" is a user application. However, it can be a 2nd-stage bootloader, which loads a user application recursively. The message `2nd boot version : 1.2` comes from a proprietary vendor 2nd-stage bootloader. I never used it or would use it in my own free time, so cannot say what's wrong with it (or with a user application using it).

Note that the reason to use 2nd-stage bootloader would be to support OTA updates. But this is an advanced, application-level feature, and I don't see why JerryScript project would be concerned with deep support of target-specific features like that (for example, if OTA support for mbedOS, Zephyr, etc. was contributed, would you accept it?). At the very least, first the application should be bootstrapped and work well using simple single-stage bootloading process. So, the best suggestion I would have is to drop two-stage bootloading build, and use just a simple single-stage setup. That won't necessarily resolve the issues, but at least will remove extra complications.

Sorry if this wasn't too helpful, but hopefully explains why I can't be of much help with this issue. Thanks.

tilmannOSG, 2016-12-05:
@pfalcon Thanks for the detailed explanation, very useful! Seems like loading JerryScript directly from the first-stage bootloader is the way to go then :)

closed 2017-11-08 by LaszloLango