jerryscript-project/jerryscript #1383: Jerry embedder should be able to decide jerry_log_level

yichoi, 2016-09-30:
IoT.js has `--jerry-memstat` option to profile JS heap usage, but it is broken now.
Jerry memory status is printed by `jerry_port_log(JERRY_LOG_LEVEL_DEBUG, ...)` function,
but under current jerryscript log level system, embedders cannot decide the log level.

yichoi, 2016-09-30:
I think there are more than one way to fix this issue.

1. Use `jerry_port_console` when printing memory status
2. Add `jerry_port_default_set_log_level()` function to jerry-port.h, and embedder will call it
3. Add `jerry_port_default_set_log_level()` function to jerry-port.h, and `jerry_init` function will call it if init flag has `JERRY_INIT_MEM_STATS` feature.
4. Or some other way?

I made a PR #1384 based on first option, but any feedback will be welcomed.

akosthekiss, 2016-09-30:
I'd opt for 4), the way jerry command line tool is doing it (as it is also an embedder of the jerryscript library). `#include "jerry-port-default.h"` in IoT.js and call `jerry_port_default_set_log_level` with debug or even trace log level if `--jerry-memstat` is given. (See https://github.com/Samsung/jerryscript/blob/master/jerry-main/main-unix.c#L212)

LaszloLango, 2016-09-30:
I agree with Ákos, #1384 is a bad idea.

yichoi, 2016-09-30:
@akosthekiss, @LaszloLango , I also have thought about that approach, but that seemed a little bit unnatural to me w.r.t multiple ports.

If I understood correctly, which level of log to print is decided in `targets/(jerry-specific target name)/*` in jerryscript. so when an embedder wants to enable memory status message for multiple ports, he/she has to:
1) find out which level of message jerry memstat log is, (it was DEBUG level)
2) check if DEBUG level log is enabled by default in each ports, (it varies among different ports)
3) enable DEBUG level log in port-specific way

And it didn't seemed to me that `jerry-port-default.h` was meant to be used for embedders,
because not every port have the equivalence of that header file.

I think it's important issue since we're targeting multiple IoT devices. 
(i.e. Memstat is important information, and we're planning to support multiple targets)
That's why I proposed opt 2) and 3), please consider this history.
And more suggestion to solve this problem would be appreciated.

akosthekiss, 2016-10-01:
@yichoi I took a closer look at the build system of IoT.js, and couldn't find a configuration where it would use a non-default port. So, including "jerry-port-default.h" might still be an option for you. (?)

As for whether it might be included or not: the original concept of the interfaces was depicted at https://github.com/Samsung/jerryscript/issues/964#issuecomment-207420477 . I.e., a port may expose API that's specific to it. In the case of the default port, these extra interfaces are defined in "jerry-port-default.h".

yichoi, 2016-10-04:
Oh, so targets/default/ is not the equivalance of other directory in targets/. Maybe I had a little confusion related with this issue: [your final observasion on this comment](https://github.com/Samsung/jerryscript/issues/964#issuecomment-207420477). I made a new PR for IoT.js to use `jerry-port-default.h` (Samsung/iotjs#422), and I will just close #1384.

yichoi, 2016-10-04:
https://github.com/Samsung/iotjs/pull/422 merged

closed 2016-10-04 by yichoi