jerryscript-project/jerryscript #1388: When I build /targets/zephyr , it shows `stdio.h: No such file or dictionary `

iamblue, 2016-10-07:
As title .

LaszloLango, 2016-10-07:
cc: @sergioamr

sergioamr, 2016-10-07:
Let me have a look, i also have to do some work on the default settings for the new zephyr since it changed a bit the api.

sergioamr, 2016-10-07:
Which version of the zephyr-sdk are you using? I cannot reproduce this issue.

iamblue, 2016-10-09:
@sergioamr 1.5.0

sergioamr, 2016-10-10:
@iamblue 1.5.0 is not the sdk. That is the Zephyr RTOS version.

If you are missing the stdio is probably because you don't have the SDK installed.
https://www.zephyrproject.org/doc/1.5.0/getting_started/getting_started.html

You have to follow the Development Environment Setup to install the SDK and the the toolchain.

This is where the SDK lives:
https://nexus.zephyrproject.org/content/repositories/releases/org/zephyrproject/zephyr-sdk/
You should have one of the latest ones. 0.8.2 for example in linux.

You can have more information about how to build zephyr here:
https://www.zephyrproject.org/

sergioamr, 2016-10-10:
@LaszloLango This seems a problem of not having the toolchain installed. 
Could we close this issue?

LaszloLango, 2016-10-10:
@sergioamr, sure. Thanks for your help.
@iamblue, feel free to reopen the issue, if you still have this problem.

closed 2016-10-10 by LaszloLango

iamblue, 2016-10-10:
@sergioamr  , I check my SDK version is 0.7.2 . I will try 0.8.2 version . Thx for your help~

sergioamr, 2016-10-10:
@iamblue If you still have issues, please feel free to contact me on my email, or join the Freenode IRC chats.

We have a #jerryscript channel there and i am normally available for chat and help during working hours (UK time).

sergioamr, 2016-10-10:
@iamblue 

You can also try our recently released zephyr.js integration that gives support to different GPIOs, sensors, etc and helps your get the maximum of your Zephyr with JavaScript. And it is a very useful project to build upon.

https://github.com/01org/zephyr.js

iamblue, 2016-10-13:
@sergioamr  zephyr.js is awesome project!

My company has some IoT platform (ex 7687 (wifi + CM4) , 7697 (wifi + BT + CM4) ). I will porting Zephyr OS and zephyr.js to our platform.