jerryscript-project/jerryscript #1392: [Question] Is there a way to free all memory allocated in JerryScript?

brianjjones, 2016-10-14:
I have a JavaScript file loaded in JerryScript that runs forever using setInterval. On occasion I need to restart this JavaScript. This code contains objects that are using jerry_set_object_native_handle, and I need to free this memory. I passed jerry_set_object_native_handle a callback to execute for releasing the handle when the JavaScript stops. If I declare the JavaScript object locally and let it go out of scope, the close callback gets called as expected. But I find this doesn't work on objects declared in the global scope, their memory just gets leaked when I restart the JavaScript.  This means that after restarting my JavaScript a couple of times I run out of memory and the system locks up. I tried calling both jerry_clear() and jerry_gc() to free any allocated memory but this still did not result in the close callback being called, and the memory remained.

Is there a way to force JerryScript to garbage collect all memory regardless of whether it is still referenced?  Basically I need my JerryScript instance to go back to square one with a completely clean slate.

Thanks.

zherczeg, 2016-10-15:
Yes, jerry_cleanup() release everything. You can call jerry_init() after that again, and the whole engine is in the initial state.Please check jerry_run_simple() as an example:

https://github.com/Samsung/jerryscript/blob/master/jerry-core/jerry.c#L239

brianjjones, 2016-10-17:
I'm afraid jerry_cleanup / jerry_init don't seem to release everything in my case.  Here's a code snippit:

var gpio = require("gpio");
var pins = require("arduino101_pins");

// This button is global.  jerry_cleanup does not release this memory.
// I'm guessing because its still being used when jerry_cleanup is called, so
// garbage collection ignores it.
var button = gpio.open({ 
    pin: pins.IO4,
    direction: 'in',
    edge: 'any'
});

// button2 is a local instance of gpio, and it does get released when jerry_cleanup is called.
// The object is no longer in use after local() is called.
function local() {   
    var button2 = gpio.open({
    pin: pins.IO5,
    direction: 'in',
    edge: 'any'
});

Any thoughts to why my global objects are not getting their memory freed by jerry_cleanup / jerry_init?

Thanks.

zherczeg, 2016-10-18:
Yes, if you manually increase the reference counter, but never release (decrease) it. In that case the GC will never release the object. Please check all objects are properly released by your application.

brianjjones, 2016-10-18:
What would we need to call to release our objects so that they get garbage collected?  Specifically we are calling jerry_set_object_native_handle(myObj, (uintptr_t)handle, closeCB) in our object, I need closeCB to get called so that it can free handle.  Also, is there a signal that we can listen for from JerryScript indicating that jerry_cleanup is occurring?

zherczeg, 2016-10-19:
jerry_release_value function is designed to decrease external reference counters. For example when you pass the ownership of a newly constructed object to the global object, you should release it. The reference counter is 1 for newly created objects. There is no signal for jerry_cleanup, but it is called by your program so you can execute any code before and after the call.

grgustaf, 2016-10-19:
@zherczeg, I've found the docs a bit lacking in terms of ownership so maybe I've misunderstood something here. If we create an object, I know its refcnt gets set to 1, but when we return it to JavaScript I thought ownership was transferred to JavaScript. This seems to be true based on Brian's example above where in the "local" case, the object is freed. If our C code still held a reference to it, it should never be freed.

But in the case where it's assigned to a global variable, it never gets freed. At least the free function we've set with set_native_handle never gets called. I think in jerry_cleanup these should all get called. The global context should effectively go out of scope, dropping refcnt of everything by one, just like what happens when exiting a local scope.

brianjjones, 2016-10-20:
I think @grgustaf explained our situation better than I did :)

LaszloLango, 2016-10-21:
@grgustaf, @brianjjones, could you privide a minimal example of the problem? Assigning something to the global context should not mean that it will leave alive after a cleanup.

zherczeg, 2016-10-21:
Perhaps it would be better for this particular case, but in general I am not sure. When a property is set, the ownership (reference) is not transferred, but a new ownership (reference) is created instead. For me this is logical, since you might want to do other things with the object, and other engines work this way. The general rule is that if you don't need a js value you need to release it (regardless of its type). The cleanup could be more clever, but if you create short living objects this way it would eat all the memory. So this error alarms that something needs to be fixed in the code.

zherczeg, 2016-10-21:
An option would be creating another set_property function, which takes ownership.

brianjjones, 2016-10-25:
Hi @LaszloLango @zherczeg  sorry for the slow reply! Here's a simple code version of what we are doing.

```
static void my_destructor_callback(uintptr_t handle)
{
    /* Free all memory here. This is never called
     * for global objects
     */
}

static jerry_value_t createMyObjectInstance (const jerry_value_t function_obj,
                                             const jerry_value_t this,
                                             const jerry_value_t argv[],
                                             const jerry_length_t argc)
{
    /* Some value check and init code here */
    
    jerry_value_t my_obj = jerry_create_object();

    /* Declare a struct and malloc memory here */
    my_struct_t *my_item = (my_struct_t *)zjs_malloc(sizeof(my_struct_t));
        
    zjs_obj_add_function(my_obj, myFunc1, "jsAPI1");
    zjs_obj_add_function(my_obj, myFunc2, "jsAPI2");
    zjs_obj_add_function(my_obj, myFunc3, "jsAPI3");

    jerry_set_object_native_handle(my_obj, (uintptr_t)my_item,
                                   my_destructor_callback);

    return my_obj;
}

/* Expose the API to the JerryScript global object */
my_obj_add_function(global_obj, createMyObjectInstance, "MyObj");
```
Now if I create both a global instance of MyObj and a local instance of MyObj in javascript, only the local one will have it's my_destructor_callback called.  The global one is not called by jerryscript_cleanup() as we'd expect.  Is this a bug?

Thanks.

zherczeg, 2016-10-26:
There can be a bug here, but I cannot figure it out because you use zjs_ and my_ prefixed functions. It would be great if you would compile jerry in debug mode. In debug, an assertion will (should) fail if not all objects are freed by cleanup. So we could check whether the callback is not called or the object is not freed.

This should be the assert:
JERRY_ASSERT (JERRY_CONTEXT (jmem_heap_allocated_size) == 0);

Or you can just change this assert to an if/print sequence.

LaszloLango, 2016-10-26:
I don't see who release `my_object` and `global_obj`. I think the problem is that you never release `my_object` or  `global_obj` and this is why the destructor is never called.

LaszloLango, 2016-10-26:
Note: We have unit test for this, so the feature works for us. I'm quite sure you are never release the global object, so the GC won't cleanup the objects assign to global.

```c
/**
 * Get global object
 *
 * Note:
 *      returned value must be freed with jerry_release_value, when it is no longer needed.
 *
 * @return api value of global object
 */
jerry_value_t
jerry_get_global_object (void)
```

Like the note said, you must release the global_object, when it is no longer needed. Usually after you assigned your own objects.

brianjjones, 2016-10-26:
AH! Calling jerry_release_value on the global_object seems to fix my problem. Now I'm seeing the callback get called for the global objects when cleanup runs.  Thanks for all the help @LaszloLango  and @zherczeg

closed 2016-10-26 by brianjjones

brianjjones, 2016-10-26:
@zherczeg @LaszloLango  one quick followup question, does calling jerry_release_value on my_object immediately delete it?  What if I want that object to continue existing in my javascript?

zherczeg, 2016-10-27:
No, currently objects are deleted only during GC. But in theory (future) it can be deleted if no reference is available to an object. From C perspective just manage the reference counter of your object. As long as it is > 0 that object cannot be deleted.

In this particular case:
my_obj = jerry_create_object(); // reference counter is 1
return my_obj; // you return with the reference, so it can be freed in by GC

However:
my_obj_other_reference = jerry_acquire_value(my_obj); // increases reference counter to 2

In practice you don't own javascript objects, but you own object references. As long as you have a reference to an object, that object cannot be deleted. When a particular reference is not needed, you can destroy it with jerry_release_value. When all references are released, the object can be freed. Note: 
my_obj_other_reference might not be equal to my_obj in C but they refer to the same content.

zherczeg, 2016-10-27:
I made a description about reference counting. Please check #1412

brianjjones, 2016-10-28:
Great, thanks for the follow up info.

grgustaf, 2016-10-28:
Yes, thanks from me too. The part I wasn't getting was that variables declared in the global context are part of the "global object" and thus will not be freed until it is. So you were right, @zherczeg, our problem was a place or two where we forgot to free the global object after using it.