jerryscript-project/jerryscript #1419: Question: any interest in an implementation of jerry-api.h targeting Emscripten?

martijnthe, 2016-11-03:
### Background:

We've been working on transpiling the application layer of the Pebble firmware + JerryScript bindings from C to JavaScript using Emscripten. The generated JavaScript targets the web browser as a host to run the transpiled application runtime, exposing the Pebble JavaScript API on a global object in the browser. Our goal with this is to enable our app developers to run the .js of their Pebble apps in the browser – just by including the app source as a `<script/>` in the HTML page.

The two big advantages for our app developers vs developing directly on real hardware:

1. The browser's built-in debugging tools can be used.
2. Much faster development cycle compared to dealing with a real device, because there is virtually no time required to transfer the code to the target.

To improve the in-browser debugging experience (1), we've decided to NOT transpile the JerryScript engine itself, but to replace the jerry-api.h implementations with JavaScript implementations (Emscripten's [EM_ASM is our friend here](https://kripken.github.io/emscripten-site/docs/porting/connecting_cpp_and_javascript/Interacting-with-code.html#interacting-with-code-call-javascript-from-native)).

### Question:

I can imagine the Emscripten-jerry-api.h implementation might also be useful for others that use JerryScript. Would you be interested to take this as a contribution?

cc @HBehrens @mjessome

zherczeg, 2016-11-07:
This sounds interesting. Am I understand correctly this would be JavaScript implementation of JerryScript?

zherczeg, 2016-11-07:
Just a question about compatibility: Jerry is an ES5 engine while the environment would likely be ES6+ where there are incompatible changes. How the implementation would be handle this? Can you share the implementation btw?

martijnthe, 2016-11-07:
> Am I understand correctly this would be JavaScript implementation of JerryScript?

Not exactly. Although in theory it's possible to transpile JerryScript to JavaScript/asm.js entirely using Emscripten (and run the JerryScript runtime inside another JavaScript runtime), this is not what we're aiming for here.

Our goal is to provide a simulation of our APIs (built on top of JerryScript) in the browser, targeting people who develop apps using our APIs. A huge advantage would be to use the browser's debugger. However, if we would transpile JerryScript entirely, this advantage would go away because the debugger would only be showing the low-level asm.js output from Emscripten.

That's why we decided to create an implementation of jerry-api.h using Emscripten's API and not actually run the JerryScript VM. This way, a call to the Jerry API would just result in executing the equivalent JavaScript directly in the browser's engine. The resulting debugging experience is much better this way.

> Just a question about compatibility: Jerry is an ES5 engine while the environment would likely be ES6+ where there are incompatible changes. How the implementation would be handle this?

Right now, it doesn't account for the incompatible changes.
Any ideas on how to handle this would be welcome.

> Can you share the implementation btw?

Yes, we're still wrapping up the work.
Once it's done, we'll put up a PR.

zherczeg, 2016-11-07:
Sorry, my question was not clear :) I was thinking about the same thing you explained, so it is basically a JavaScript engine written in JavaScript, which has a similar API as JerryScript. We are definitely interested in the PR, this could greatly accelerate prototyping JS code.

zherczeg, 2017-02-03:
Is this work is still in-progress?

martijnthe, 2017-02-03:
Hi @zherczeg, sorry for the delay. I haven't had much time to work on this due to the transition from Pebble to Fitbit. I'll try to make some time to address your comments and feedback. Thanks for the reminder ;)

Related PR: https://github.com/jerryscript-project/jerryscript/pull/1432

rerobika, 2019-10-25:
The related #1432 and #1901 have already been closed, so I think we can finally close this issue.

closed 2019-10-25 by rerobika

slaff, 2019-10-26:
If someone is interested we have updated snapshot compiler running with latest Emscripten + latest JerryScript at this repository:

https://github.com/attachix/jerryscript/tree/feature/emscripten-jerryscript-v2

martijnthe, 2019-10-27:
@slaff nice to see our efforts live on :)

slaff, 2019-10-27:
> nice to see our efforts live on :)

@martijnthe I still find the idea quite useful :) Nowadays the generated and compressed JS and WASM  are almost half the size of the pure JS file that was generated in the Emscripten SDK 1.37.40 days. Not to mention that on modern browsers also the execution is faster.