jerryscript-project/jerryscript #1424: ecma_builtin_global_object_print should add carriage return (\r)

janjongboom, 2016-11-10:
On many terminal / serial monitor applications (incl. screen on OS/X) it's required to use both a carriage return (`\r`) and a newline (`\n`) to properly create newlines. Is there a reason why `ecma_builtin_global_object_print` only does a newline?

I believe changing the call to `jerry_port_console ("\n");` into `\r\n` or `\n\r` should be the right thing to do.

akosthekiss, 2016-11-10:
I'm not sure that the engine should be dealing with the `\n\r`, `\r\n`, `\r`, or  `\n` translations. I found the following explanation (somewhere out in the wild) quite descriptive:

"In file I/O, in text mode, upon reading the bytes of the input file (and stdin), depending on the environment, when bytes with the value(s) of 10 (Unix), 13,10 (Windows), 13 (Old Mac??) and other variations are translated in to a '\n'. Upon writing a file (or stdout), the reverse translation occurs. Note: File I/O in binary mode makes no translation."

To me, the key point is "depending on the environment". From the perspective of C, `\n` represents EOL, not necessarily the byte 10 only. So, if the port needs any translation, it should apply it in `jerry_port_console` (or in the library functions that get called by `jerry_port_console`, if any).

PS: I've just tested jerry on OSX, all line feeds happen correctly.
```
$ tools/build.py --clean
$ ./build/bin/jerry --version
Version: 1.0 (9f556e1)
$ ./build/bin/jerry
jerry> print('foo\nbar')
foo
bar
undefined
jerry>
```

janjongboom, 2016-11-11:
@akosthekiss 

> depending on the environment

Sure, I can live with that. Seen this on devices where the logging information is sent over serial, but if we can fix this in `jerry_port_console` that'd work for me too.

closed 2016-11-11 by janjongboom