jerryscript-project/jerryscript #1426: mbed OS 5 port should add carriage return (`\r`) in jerry_port_console

janjongboom, 2016-11-11:
When calling `print` from JS code only a newline is added after the log message. This does not give a full line break on many serial monitors (like `screen` on OS/X and Linux). We should augment `jerry_port_console` to add a carriage return at the end of the line.

janjongboom, 2016-11-11:
I'll write a patch for this.

akosthekiss, 2016-11-11:
A quick question before patching the code base: have you tried configuring the serial monitor to do the LF -> CRLF translation?
Some discussion on this: http://stackoverflow.com/questions/7812142/how-to-toggle-cr-lf-in-gnu-screen

janjongboom, 2016-11-12:
@akosthekiss Sure, that'd work, but... I'd rather have it display correctly for all users when they start debugging.

closed 2016-11-28 by LaszloLango