jerryscript-project/jerryscript #1434: Automatic benchmark does not work

LaszloLango, 2016-11-15:
The public benchmark is not refreshing: http://jerryscript.net/benchmark/benchmark.html
The last data is from the end of august. Do we want to keep it? Any plan to make it work again? IMHO it is not too informative with such a small subset of Sunspider.

LaszloLango, 2016-11-15:
cc: @wateret

tilmannOSG, 2016-11-23:
@LaszloLango Improving our performance tracking is definitely on our TODO list for the coming months. The current setup has various issues (commits performance data into the main repository, small benchmark coverage) and I don't think we should revive it as is. We probably need to have a more detailed discussion about where we want to go in terms of performance tracking infrastructure.

LaszloLango, 2016-11-23:
@tilmannOSG, I agree. A new benchmark system would be the best option. I think the best way is to put it into the cloud, but don't know any service that fits our needs. I'd be surprised if there is no such service on the internet. If we put it into the cloud, then the benchmark system will be independent from any contributor and their location.
Other solution could be to improve our internal benchmark system (that we are using on performance and memory related PRs) and make it public.

Furthermore we should consider to write a new benchmark system that covers the standard better than Sunspider and rely more on embedded systems.

lemmaa, 2016-11-23:
@LaszloLango, I have just seen this issue. I am sorry that the benchmark system has been shut down due to internal circumstances. I believe you will build a better system.
One thing, I think the existing benchmark data is very important to the history of Jerry script. I hope it is well preserved. :)

zherczeg, 2017-01-31:
Closing due to inactivity. Please reopen if needed.

akosthekiss, 2017-10-30:
The new benchmark system has been available for a while now at https://jerryscript-project.github.io/jerryscript-test-results/ . Closing.

closed 2017-10-30 by akosthekiss