jerryscript-project/jerryscript #1437: jerry save-snaphot-for-global hangs

nkolban, 2016-11-17:
I build Jerry on Linux but when I run:

```
./jerry --save-snapshot-for-global t.js output.snp
```

The program just hangs/loops.  If there are additional diagnostics I can run, please let me know.

LaszloLango, 2016-11-17:
We are testing snapshot feature for every commit in Travis CI and works fine in our test environment.
Could you provide more information? What is in "t.js"? What is the version of JerryScript you are working on? What is your current architecture, x86_64?

nkolban, 2016-11-17:
Thanks for the response.   Here are the details:

I am building on Ubuntu on an x86-64.

Jerry report as v1.0 (ffaca58)

It built 100% cleanly from a github extract I performed on 2016-11-16 from master.

I tried to run against a variety of JavaScript source files.  My current source file is minimal with:

```
print("hello");
```

zherczeg, 2016-11-17:
I think the parameter order is worng. Swap t.js and output.snp.

nkolban, 2016-11-17:
Aha!!  That makes sense ...  now when I run I get the following:

```
$ ./jerry --save-snapshot-for-global output.snp t.js
Script Error: Error
```

My goodness ... we sure love mystery in this project :-)

LaszloLango, 2016-11-18:
IMHO, wrong parameter order should not hang the engine. We must fix this.

zherczeg, 2016-11-18:
You need to compile in debug mode or enable error messages to get nice output.

zherczeg, 2016-11-18:
(JERRY_ENABLE_ERROR_MESSAGES define must be enabled)

PR #1442 (2016-11-21): Fix fopen call on non-existing files.
The main implementation of unix platform hangs, if the engine was
built with jerry-libc. Open system call returns with a negative
error code if cannot open the file, but fopen must return NULL
when error happens.

Fixes #1437.

JerryScript-DCO-1.0-Signed-off-by: László Langó llango.u-szeged@partner.samsung.com
commit 4a0f78bc (2016-11-21): Fix fopen call on non-existing files. (#1442)

The main implementation of unix platform hangs, if the engine was
built with jerry-libc. Open system call returns with a negative
error code if cannot open the file, but fopen must return NULL
when error happens.

Fixes #1437.

JerryScript-DCO-1.0-Signed-off-by: László Langó llango.u-szeged@partner.samsung.com
closed 2016-11-21 by LaszloLango