jerryscript-project/jerryscript #1475: ArrayBuffer needs regression tests

tilmannOSG, 2016-12-07:
As discussed in #1467, we need to add a set of regression tests to have some basic test coverage of the ArrayBuffer implementation.

tilmannOSG, 2016-12-07:
@jiangzidong Created this issue so we don't forget about the tests :)

jiangzidong, 2016-12-07:
yes, it is a good place to ask and discuss test-related questions.

jiangzidong, 2016-12-08:
@LaszloLango We have 2 test folders: "tests/jerry" and "tests/jjerry-test-suite". What is the difference betweem them?

LaszloLango, 2016-12-08:
@jiangzidong, good question. :) It is quite legacy, not sure what was the intention of this. I think the jerry-test-suite was designed to use it instead of the official test262 test suite. "tests/jerry" is mostly regression tests. If we fix a bug, the related regression test must be put there. But there are some general tests too in that folder.

commit fb2edd85 (2016-12-09): Add ArrayBuffer regression tests (#1477)

Add a set of tests to ensure basic coverage of the ArrayBuffer built-in. Support for ArrayBuffer was added in #1467.

Closes #1475.

JerryScript-DCO-1.0-Signed-off-by: Zidong Jiang zidong.jiang@intel.com
closed 2016-12-09 by tilmannOSG