jerryscript-project/jerryscript #1511: Problem with building jerryscript on mbed os 5

limjunsung, 2017-01-04:
I try the tutorial below to build jerryscript to an mbed os 5 target.
https://developer.mbed.org/blog/entry/Using-mbed-libraries-with-JerryScript/

In the Setting up section of the tutorial, you can see:

<------------------------
Setting up

The following instructions guide you through the process of setting up an environment where you can run JavaScript on mbed OS:

Open a terminal and navigate to the folder where you cloned the JerryScript repository.
Run cd targets / mbedos5.
Run make getlibs - this will pull in mbed OS and all related repositories.
Run pip install -r ./tools/requirements.txt - this will install Python modules required by the build scripts.
------------------------->
The problem is that when I run the 'make getlibs' command, the following error message appears.

<------------------------
[mbed] ERROR: Could not find mbed program in current path "/ home / parallels / jerryscript / targets / mbedos5".
[mbed] ERROR: Change the current directory to a valid mbed program or use the 'global' option to set global configuration.

mbed config root.
[mbed] ERROR: Could not find mbed program in current path "/ home / parallels / jerryscript / targets / mbedos5".
[mbed] ERROR: Change the current directory to a valid mbed program or use the 'global' option to set global configuration.

make: *** [.mbed] Error 255
------------------------->

So I opened the Makefile and looked at the getlibs configuration section.

<------------------------
getlibs: .mbed

.mbed:
mbed config root.
mbed toolchain GCC_ARM
mbed target $ (BOARD)
mbed deploy
------------------------->

I would like to know what to do if the above error occurs.

Thanks in advance

LMESTM, 2017-01-04:
I confirm that I'm facing the same issue here.

zherczeg, 2017-01-04:
@janjongboom @thegecko any idea?

LMESTM, 2017-01-04:
Note, I think I could somehow workaround the issue by issuing:
`mbed new -vv --create-only .` 

but then the call to make getlibs says there's nothing to be done 
`$ make getlibs
make: Rien ▒ faire pour ▒ getlibs ▒.` 

So I had to manually
 `mbed config root .`
        `mbed toolchain GCC_ARM`
        `mbed target $(BOARD)`
        `mbed deploy`

and now finally 
`make BOARD=... `
seemed to work fine 

Note that I think I hit the same issue when follwoing npm/gulp based example here:
https://github.com/janjongboom/mbed-js-repl-example

thegecko, 2017-01-04:
I'm away from a PC right now, could you try creating a file named `.mbed` in the project root;

```bash
$ touch .mbed
```

 and seeing if it fixes the issue?

janjongboom, 2017-01-04:
Confirmed issue with mbed CLI 1.0.0... Does not build unless a `.mbed` file is present in the directory. Writing a patch for our Makefile right now.

janjongboom, 2017-01-04:
Fixed in #1512.

limjunsung, 2017-01-05:
LGTM Thank you so much!!!!

limjunsung, 2017-01-05:
It seems to be working, but the error message still appears...

$ make getlibs

[mbed] ERROR: Could not find mbed program in current path "/Users/limjunsung/jerryscript/targets/mbedos5".
[mbed] ERROR: Change the current directory to a valid mbed program, set the current directory as an mbed program with 'mbed config root .', or use the '--global' option to set global configuration.
echo 'ROOT=.' > .mbed
mbed config root .
[mbed] . now set as default root in program "mbedos5"
mbed toolchain GCC_ARM
[mbed] GCC_ARM now set as default toolchain in program "mbedos5"
mbed target
[mbed] No default target set in program "mbedos5"
mbed deploy
[mbed] WARNING: Program "mbedos5" in "/Users/limjunsung/jerryscript/targets/mbedos5" does not use source control management.
[mbed] WARNING: To fix this you should use "mbed new ." in the root of your program.
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at rev #d5de476f74dd

PR #1512 (2017-01-05): target: mbedos5: mbed CLI refuses to build if no .mbed file present
With the newest release of mbed CLI (1.0.0) it error's when no .mbed file is present in the project root. This breaks builds for JerryScript on mbed OS 5. With this patch we create this file and builds succeed again. We cannot use `mbed new` because it also creates a new git repository in targets/mbedos5 folder which is not what we want.

JerryScript-DCO-1.0-Signed-off-by: Jan Jongboom janjongboom@gmail.com

/cc @thegecko @LMESTM @zherczeg @limjunsung

Fixes #1511
closed 2017-01-05 by LaszloLango