jerryscript-project/jerryscript #1514: Question: when I use some js functions, jerry_run_simple() return false

limjunsung, 2017-01-06:
I want to put the jerryscript engine on the target board and interactively send the JavaScript code and execution results between the target board and the PC terminal via serial communication.

I am using the NUCLEO_F446RE board, and I have succeeded in building the jerryscript engine to target mbed os 5.
Then I followed Step 1 of the tutorial below and got 'Hello World!' output.
Https://github.com/janjongboom/jerryscript/blob/master/docs/03.API-EXAMPLE.md

<main.cpp>
#include <string.h>
#include "mbed.h"
#include "jerry-api.h"

Serial pc(USBTX, USBRX)

int main (int argc, char *argv[])
{
  const jerry_char_t script[] = "print ('Hello, World!');";
  size_t script_size = strlen ((const char *) script);

  bool ret_value = jerry_run_simple (script, script_size, JERRY_INIT_EMPTY);

  return (ret_value ? 0 : 1);
}

The problem is that when I use function codes such as setInterval(), DigitalOut() instead of the print() function, jerry_run_simple() return false.

Ex.
const jerry_char_t script [] = "setInterval ( function () {print ('a') }, 1000);

I want to know why this is happening and what should I do.

Thanks in advance. #

zherczeg, 2017-01-06:
SetInterval and friends are provided by the environment, not the JS engine. As far as I know mbed5 provides these functions. Perhaps @janjongboom @thegecko could help.

janjongboom, 2017-01-06:
@limjunsung I haven't looked at the underlying issue yet, but does it work properly when you run `jerry_run`? I'm using that for the REPL I was building and that worked without issues...

See https://github.com/ARMmbed/mbed-js-repl/blob/master/repl/repl.h#L248

LaszloLango, 2017-01-06:
There are only the standard builtins available when you are using `jerry_run_simple`. This function is init a new environment at start and clean it up at the end.

closed 2017-01-19 by limjunsung