jerryscript-project/jerryscript #1519: checking instanceof against a native external function gives TypeError instead of true/false

grgustaf, 2017-01-10:
Here's a transcript with bin/jerry that demonstrates the issue:

```
jerry> function foo() { }
undefined
jerry> typeof foo
function
jerry> typeof assert
function
jerry> null instanceof foo
false
jerry> null instanceof assert
Script Error: TypeError
jerry> bar = new foo()
[object Object]
jerry> bar instanceof foo
true
jerry> bar instanceof assert
Script Error: TypeError
```

I'm wanting to write a JS constructor in C, and I think it would probably work except for this problem. What happens here to assert also happens to my function.

zherczeg, 2017-01-11:
I made fix in #1520. However, you still need to manually create prototypes. See the regression tests.

grgustaf, 2017-01-11:
OK I can confirm the patch does fix the problem as stated. Interestingly, it doesn't seem to fix the same problem for an external function *I* add myself though. I've done basically this:

```c
jerry_value_t myfunc(const jerry_value_t ...) {
    return jerry_create_undefined();
}
...
jerry_value_t global = jerry_get_global_object();
jerry_value_t func = jerry_create_external_function(myfunc);
jerry_value_t name = jerry_create_string("EmptyObject");
jerry_set_property(global, name, func);
...
```

Then in JavaScript I see that `null instanceof EmptyObject` still gives me a TypeError. I can create an object `foo = new EmptyObject()` but `foo instanceof EmptyObject` also gives me a TypeError.

So apparently this is a slightly different problem than with assert, but this is the one I'm really looking to solve. :) I can open a new issue if you prefer.

grgustaf, 2017-01-11:
Now I see what you meant about manually setting prototypes. The patch fixes `null instanceof assert` but `({} instanceof assert)` still gives TypeError unless you set a prototype.

But even if I set a prototype: `EmptyObject.prototype = {}` in JS, checking `null instanceof EmptyObject` still gives TypeError. Also, if I try to set the prototype on `func` in C, it the .prototype still shows up as undefined in JS, and still gives a TypeError, though `jerry_set_prototype` returns success.

zherczeg, 2017-01-12:
> Now I see what you meant about manually setting prototypes. The patch fixes null instanceof assert but ({} instanceof assert) still gives TypeError unless you set a prototype.

This is the expected.

I will need to check the rest. There are lot of rules here where you need to return with type error or false. Yes, instanceof is expected to return with type error in many cases.

zherczeg, 2017-01-12:
The Instanceof operator

http://www.ecma-international.org/ecma-262/5.1/#sec-11.8.6

5. If Type(rval) is not Object, throw a TypeError exception
6. If rval does not have a [[HasInstance]] internal method, throw a TypeError exception.

It seems only two types has [[HasInstance]] internal property

http://www.ecma-international.org/ecma-262/5.1/#sec-15.3.5.3
http://www.ecma-international.org/ecma-262/5.1/#sec-15.3.4.5.3

2. Let O be the result of calling the [[Get]] internal method of F with property name "prototype".

The prototype property and prototype of an object is two different things.

    var o = {}
    o.prototype = { a:5 }
    Object.setPrototypeOf(o, { a:6 })
    o.prototype.a // =5
    o.a // =6

The C API implementation of set_prototype implements the Object.setPrototypeOf operation.

At first sight the implementation looks correct. If I understand correctly null is not an object, it represents the absence of an object.

grgustaf, 2017-01-12:
> 5. If Type(rval) is not Object, throw a TypeError exception

`rval` is the "ShiftExpression" though, not the "RelationalExpression". In my example, `rval` is an Object (the EmptyObject function), it's the `lval` that I'm making null in my example.

But I guess it could break on the HasInstance test, I don't understand those internal details.

Just looking from the perspective of the JS user/app, I can't tell the difference between a JS function, a built-in function like assert, and a function created by jerry_create_external_function. They all show up as typeof "function". So it seems like instanceof should work identically on each of them: If I give it a function on the right, it should give me true or false.

What I'm wanting to do is create a valid JS constructor from the C API, which will be able to create new objects that can be checked against that constructor with instanceof, as they can with constructors written in JS. Is there a way to do that?

(I've been modifying this comment all day, I'm just going to post it now. :D)

grgustaf, 2017-01-13:
Ah, thanks to your comments I think I resolved my issue. You pointed out the difference between .prototype and setPrototypeOf/jerry_set_prototype. It turns out if if I call jerry_set_property() to set "prototype" property, it works.

The one remaining weirdness there is that if you print out "EmptyObject.prototype" in JS it shows up as undefined, even though there really is a prototype set. And if you set it from JS instead with EmptyObject.prototype = foo, then it *will* show up when printed.

grgustaf, 2017-01-13:
One last thing I realized. If I don't set "prototype" property on my external function, then it fails the instanceof test with TypeError like we discussed above. Setting it fixes that so that instanceof will work and return true or false as appropriate.

But that does make me wonder, shouldn't an external function have a default "prototype" property of Object() like I think it does when you make a function within JS?

zherczeg, 2017-01-13:
Again, I want to emphasize that the prototype of an object and a prototype property is two different things. Try this in a browser console:

    ({} instanceof Math.sin)
    Firefox: TypeError: 'prototype' property of Math.sin is not an object

This is not a JerryScript specific behaviour, this is the ES standard. Unlike objects, functions has two prototypes (I know this is confusing), one is the real prototype which is mandatory and another is a prototype property which is optional. Certain ES functions uses the first, others uses the second. Another example:

    var f = function() {}
    delete f.prototype
    ({} instanceof f)
    Firefox: TypeError: f.prototype is not a function

So no, external functions and standard built-ins (such as Math.sin) has no default prototype property, although they have a real prototype, which is null. I don't want to change this, since for me external functions are the same as built-ins with a fancy name. But their purpose is exactly the same.

> The one remaining weirdness there is that if you print out "EmptyObject.prototype" in JS it shows up as undefined, even though there really is a prototype set. And if you set it from JS instead with EmptyObject.prototype = foo, then it will show up when printed.

I don't really understand this. If you set a property with jerry_set_property, it must be visible in ES.

grgustaf, 2017-01-14:
OK, I see you're right about that. `instanceof` on a built-in like "print" or "Math.sin" does fail with TypeError in Chrome as well. Sorry, I know I saw that before, I just confused myself with the dozens of things I tried.

It does remain strange that I can't see the "prototype" I set with jerry_set_prototype. But that is the critical line that lets me define a new object constructor from C and have it work as expected with `instanceof`.

My real problem is now solved, even without your patch. For what it's worth, here's what I was [trying to accomplish](https://github.com/01org/zephyr.js/pull/599/files#diff-b370cf679111fa315eabf0716b012997R64) (scroll up a little to line 64), which is perhaps foolishly reimplementing part of what JerryScript doesn't include in the minimal build. (But also a convenient way to add our own errors.)

zherczeg, 2017-01-16:
> It does remain strange that I can't see the "prototype" I set with jerry_set_property.

That is strange. Especially that "instanceof" can use it. Do you have an example?

commit 4b5cf8ad (2017-01-16): HasInstance should accept the prototype of all functions. Fixes #1519. (#1520)

The regression-test-issue-736.js is also changed, since it expects
throwing an error, which does not happen anymore because the "x"
variable is undefined.

JerryScript-DCO-1.0-Signed-off-by: Zoltan Herczeg zherczeg.u-szeged@partner.samsung.com
closed 2017-01-16 by tilmannOSG