jerryscript-project/jerryscript #1530: Question: I have a linking problem

limjunsung, 2017-01-17:
Hi, all.

I followed the example here(https://github.com/janjongboom/jerryscript/tree/the-answer)

I cloned that repository and got the following error when I ran make file.

```
Scan: FEATURE_STORAGE
Scan: 
Scan: mbed
Scan: env
Compile: launcher.cpp
Link: mbedos5
./BUILD/NUCLEO_F446RE/GCC_ARM/jerryscript-mbed/jerryscript-mbed-launcher/source/setup.o: In function `jsmbed_js_load_magic_strings()':
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x6c): undefined reference to `jsmbed_js_magic_string_count'
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x70): undefined reference to `jsmbed_js_magic_string_lengths'
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x74): undefined reference to `jsmbed_js_magic_strings'
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x78): undefined reference to `jsmbed_js_magic_string_values'
collect2: error: ld returned 1 exit status
[ERROR] ./BUILD/NUCLEO_F446RE/GCC_ARM/jerryscript-mbed/jerryscript-mbed-launcher/source/setup.o: In function `jsmbed_js_load_magic_strings()':
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x6c): undefined reference to `jsmbed_js_magic_string_count'
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x70): undefined reference to `jsmbed_js_magic_string_lengths'
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x74): undefined reference to `jsmbed_js_magic_strings'
setup.cpp:(.text._Z28jsmbed_js_load_magic_stringsv+0x78): undefined reference to `jsmbed_js_magic_string_values'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /home/parallels/dev2/jerryscript/targets/mbedos5/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F446RE --source . --source ../../ --build ./BUILD/NUCLEO_F446RE/GCC_ARM -j0 -D CONFIG_MEM_HEAP_AREA_SIZE=(1024*16)" in "/home/parallels/dev2/jerryscript/targets/mbedos5"
---
make: *** [all] Error 1
```

Any help will be greatly appreciated.

tilmannOSG, 2017-01-17:
Adding @janjongboom and @thegecko as this seems to be related to the mbed OS port.

janjongboom, 2017-01-17:
@limjunsung That branch is quite outdated... Can you run through the 'TheAnswer' example with https://github.com/armmbed/mbed-js-example? The C++/JS bindings will be exactly the same so you can follow along, but mbed-js-example is constantly updated.

Just built it for NUCLEO_F446RE and it builds & links fine...

limjunsung, 2017-01-17:
@janjongboom Thank you for your reply. I have confirmed that it works well. Actually, I really want to make real-time javascript interpretor running on mbed os 5. So, I tried to run the example that you wrote (https://github.com/janjongboom/mbed-js-repl-example), but it shows error message below whenever I run the command 'gulp --target=NUCLEO_F446RE'.

```
Compile [ 99.2%]: jerry_port_mbed.c
[Warning] jerry_port_mbed.c@61,7: implicit declaration of function 'strlen' [-Wimplicit-function-declaration]
Compile [ 99.5%]: stm_spi_api.c
Compile [ 99.7%]: test_env.cpp
Compile [100.0%]: repl_js.cpp
[Warning] repl.h@111,49: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
[Warning] repl.h@133,49: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
[Warning] repl.h@189,49: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
[Warning] repl.h@207,28: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
[Warning] repl.h@212,33: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
[Warning] repl.h@257,45: variable length array 'buffer' is used [-Wvla]
Link: mbedos5
/Users/junsunglim/gcc-arm-none-eabi-6_2-2016q4/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: ../../../out/NUCLEO_F446RE/mbedos5.elf section `.text' will not fit in region `FLASH'
/Users/junsunglim/gcc-arm-none-eabi-6_2-2016q4/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 18196 bytes
collect2: error: ld returned 1 exit status
[ERROR] /Users/junsunglim/gcc-arm-none-eabi-6_2-2016q4/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: ../../../out/NUCLEO_F446RE/mbedos5.elf section `.text' will not fit in region `FLASH'
/Users/junsunglim/gcc-arm-none-eabi-6_2-2016q4/bin/../lib/gcc/arm-none-eabi/6.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 18196 bytes
collect2: error: ld returned 1 exit status

[mbed] ERROR: "python" returned error code 1.
[mbed] ERROR: Command "python -u /Users/junsunglim/dev/mbed-js-repl-example/build/jerryscript/targets/mbedos5/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_F446RE --source . --source ../../ --source /Users/junsunglim/dev/mbed-js-repl-example/node_modules/mbed-js-repl/. --source ../../../source --build ../../../out/NUCLEO_F446RE -j0 -D CONFIG_MEM_HEAP_AREA_SIZE=(1024*16)" in "/Users/junsunglim/dev/mbed-js-repl-example/build/jerryscript/targets/mbedos5"
---
make[1]: *** [all] Error 1
make: *** [all] Error 2
events.js:160
      throw er; // Unhandled 'error' event
      ^

Error: Command `make BOARD=NUCLEO_F446RE EXTRAS=/Users/junsunglim/dev/mbed-js-repl-example/node_modules/mbed-js-repl/.` exited with code 2
    at ChildProcess.handleSubShellExit (/Users/junsunglim/dev/mbed-js-repl-example/node_modules/gulp-run/command.js:166:13)
    at Object.onceWrapper (events.js:290:19)
    at emitTwo (events.js:106:13)
    at ChildProcess.emit (events.js:191:7)
    at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
```

Could you give me a advice that solves the problem?

janjongboom, 2017-01-17:
@limjunsung For some reason the REPL uses 150K more flash, which makes it fail linking on most targets (except K64F)... Not sure why it uses so much. Hope to spend some time on it this week. The map file is a lot bigger, so would be good to dig through there.

janjongboom, 2017-01-17:
Quick inspection: probably because it relies on libstdc++ and that uses a lot of flash. Let's see if that can be fixed.

janjongboom, 2017-01-17:
So stringstream has a much larger footprint than I thought. Uses 140K of flash... Let's see if I can rewrite without stringstreams.

janjongboom, 2017-01-17:
@limjunsung I wrote a patch here: https://github.com/janjongboom/mbed-js-repl/pull/1, now it compiles and links against NUCLEO_F446RE.

limjunsung, 2017-01-17:
@janjongboom It works well!!!! Thank you soooooo much!!

closed 2017-01-18 by LaszloLango